ARMmbed/mbed-os #4759: Build error occurs by exporting mbed classic sample to GCC (ARM Embedded) toolchain on mbed online compiler

TomoYamanaka, 2017-07-14:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

As exporting **mbed classic sample(e.g. mbed-blinky)** to GCC (ARM Embedded) toolchain on mbed online compiler, the below build error will occurs.
This error occurs by using mbed classic **rev.145 or later**.

```
../mbed/TARGET_RZ_A1H/TOOLCHAIN_GCC_ARM/system_MBRZA1H.o: In function `InitMemorySubsystem':
system_MBRZA1H.c:(.text.InitMemorySubsystem+0x24): undefined reference to `__v7_all_cache'
collect2.exe: error: ld returned 1 exit status
```

I consider that cause of this error is the lack of `__v7_all_cache` in GCC only.
Before rev.144, it's function has been implemented. 
But in rev.145, it seems that the substance of it's function does not exist because `cache.S` has been deleted, as a result, the above error occurs.

0xc0170, 2017-07-14:
I assume its RZ_A1H target. who and when was cache.S deleted?

cc @theotherjimmy

theotherjimmy, 2017-07-14:
I don't have the answer to that question @0xc0170

theotherjimmy, 2017-07-14:
CMSIS 5 update? seems to affect all releases >= 5.5.0

0xc0170, 2017-07-17:
Thanks Jimmy.

@TomoYamanaka For gcc, this function is declared as extern, where was the definition of this function? What we are missing here as this is local issue, what is your GCC ARM version?

toyowata, 2017-07-19:
@0xc0170 
The ```cache.S``` was in here, but removed by CMSIS5 update:
https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.4/cmsis/TOOLCHAIN_GCC/TARGET_CORTEX_A

0xc0170, 2017-07-19:
Thanks @TomoYamanaka . I'll send PR shortly, adding this file back

commit 0eb4cf0d (2017-08-14): Add cortex-a cache file

Fixes #4759

This was removed during cmsis5 update. Here is original file:
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.4/cmsis/TOOLCHAIN_IAR/TARGET_CORTEX_A/cache.S and
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.4/cmsis/TOOLCHAIN_GCC/TARGET_CORTEX_A/cache.S
closed 2017-08-14 by theotherjimmy