ARMmbed/mbed-os #4763: NucleoF446ze CAN Configuration bug

mr63, 2017-07-17:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NucleoF446ze 

**Toolchain:**
ARM

**mbed-cli version:** 
   146:22da

**Expected behavior**
Set variation of CAN for sync.

**Actual behavior**
When value is set it is Masked out because of wrong bit mask and has no effect.  Leaving the variation to 0.
**Steps to reproduce**

Testing from one internal channel to another has no effect as they both use the same internal clk source and are always in sync.  You must connected to external CAN source and will see that most packets are not excepted.

The problem is found in the can_api.c File line 184
inside of the can_speed function.

when psjw is set it is masked out with the wrong mask value the mask value should be set to:
0x03000000.

Thank You,
Mark

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2017-07-17:
cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-07-21:
Hi
Please, be free to send a pull request.
Thx

bcostm, 2017-07-24:
Fixed in PR #4801

bcostm, 2017-08-16:
Can you please close this issue ? The PR has been merged. Thanks.

ST_TO_BE_CLOSED

closed 2017-08-17 by 0xc0170