ARMmbed/mbed-os #4769: STM32 F303 I2CSlave: only standard mode is working

adbam, 2017-07-18:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
Nucleo_F303RE

**Toolchain:**
GCC_ARM

**Toolchain version:**
from ac6: gcc-arm-none-eabi-5_4-2016q3-20160926-win32

**mbed-cli version:** 
1.1.1

**Expected behavior**
I2cSlave should be able to work with clockrates up to 1MHz (I2C Fm+)

**Actual behavior**
I2C Standard mode works, Fast mode and Fast mode plus doesn't work.

**Steps to reproduce**
main.cpp
```
#include "mbed.h"

DigitalOut led1(LED1);
I2CSlave slave(PB_9, PB_8);

int main() {
	char buf[10];

	slave.frequency(1000000);
    slave.address(0x20);
    while (true) {
        int i = slave.receive();
        switch (i) {
            case I2CSlave::ReadAddressed:
                //slave.write(msg, strlen(msg) + 1); // Includes null char
                break;
            case I2CSlave::WriteGeneral:
                slave.read(buf, 10);
                break;
            case I2CSlave::WriteAddressed:
                slave.read(buf, 10);
                led1.write(buf[0]);
                break;
            default: break;
        }
        for(int i = 0; i < 10; i++) buf[i] = 0;    // Clear buffer
    }
}
```
Master sends 0x10 (7bit address) and 0x01 to enable LED

Standard Mode (100kHz)
[image]

Fast Mode (400kHz)
[image]

Fast Mode Plus (1MHz)
[image]

0xc0170, 2017-07-18:
cc @bcostm @adustm @LMESTM @jeromecoutant

LMESTM, 2017-07-18:
@adbam thanks for reporting the issue with diagrams. 
I have a few questions to ask in a first step:
- what is the mbed version being used ? 
- what is the I2C master that you are using ? 
- and does the I2C master support clock stretching ? 
Because MBED I2C slave is a SW implementation, based on a polling loop which is visible in the main, it takes some time before the master request is acknowledge. This actually requires the I2C slave to take some time before releasing the SCL (i.e. I2C slave here applies some clock stretching technique as my current understanding of clock stretching from here https://www.i2c-bus.org/clock-stretching/)

I think this is visible in your diagrams where your analyzer show the "ACK" only when the clock. But this looks to me as if the master is not considering this clock stretching .. the SDA line should only go UP after the 8th clock cycle, as in the 100kHz example. In your FM and FM+ diagrams, the SDA line goes UP after a fixed amount of time, not after the 8th SCL clock cycle

adbam, 2017-07-19:
Hi LMESTM, thanks for looking into this. 

To answer the questions:
- I use mbed_lib_rev146 which was checked out with mbed-cli
- The master is a FT2232
- I run some tests today and it seems as if clock stretching is not supported by FT2232

Mentioning unsupported clock stretching lead me to the right path. Reinterpreting the diagrams it makes complete sense now and it seems as if the master simply continues to send, but the slave is not ready yet.

LMESTM, 2017-07-19:
@adbam yes this is also my understanding ... so I'm affraid I won't be able to solve your problem. As mentioned, I2C slave being fully SW based, moreover using a polling loop, we won't be able to cope with very high rate without clock stretching support.  
Thanks ion advance to close the issue when you think all questions are answered.

closed 2017-07-20 by adbam