ARMmbed/mbed-os #4770: Rising and falling edge is not properly counted by IRQ handler.

akrnvm, 2017-07-18:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
Custom board

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 
1.1.1

**Expected behavior**

interrupt should count properly

**Actual behavior**

interrupt is not counting properly

**Steps to reproduce**
----------------------------------------------------------------
Description:
In MbedOS 5.4.7, 
When I'm giving pwm out to interrupt pin, getting expected count (expected count:100).
Count : Rise count = 101 and Fall count = 101.

But the same application in MbedOS 5.5.2, count is not proper (expected count:100). 
Count : Rise count = 13 and Fall count = 13.

Note:- While debugging in keil I'm getting expected value.

Below is the application code.

```
#include "mbed.h"

int rise_count;
int fall_count; 

void rise(void)
{
    rise_count++;
}

void fall(void)
{
    fall_count++;
}

PwmOut pwm(MBED_PWMOUT0);
InterruptIn sw(INT0);
Serial pc(USBTX, USBRX);

int main(void)
{
    fall_count = 0;
    rise_count = 0;
    
    sw.rise(rise);
    sw.fall(fall);
    
    pwm.period(0.030);
    pwm.write(0.5f);
    
    wait(3);
    pc.printf("\r\nRise = %d, Fall = %d, And Excepted = %d\r\n", rise_count, fall_count, 100);
    while(1);
}
```
-----------------------------------------------------------------

## Question
How to resolve this issue?

0xc0170, 2017-07-18:
As this is custom board, we should look at upper layers to find out what is causing this difference. I can see you are calling wait(). Have you tested release and also debug profile ?

I dont recall Serial/InterruptIn neither PwmOut changes. The main change between those 2 tested versions is RTX update, sleep call in the idle loop for the release build. Thus would like to see if debug profile does the same.

akrnvm, 2017-07-19:
Thank you for the comment,
I have tested in debug mode and its working fine.
I have missed some setting in sleep function.
After fixing, test cases is passing .

closed 2017-07-19 by akrnvm