ARMmbed/mbed-os #4771: "Cortex-M4F" Serial Output is not printing

pranavshandilya, 2017-07-18:
### Description
- Type: Bug 

- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
Custom Board

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 

**Expected behavior**
Serial Out should come

**Actual behavior**
Serial Output is not printing

**Steps to reproduce**

----------------------------------------------------------------

When I changed from "Cortex-M4" to "Cortex-M4F" for Custom Target(Core:- Cortex-M4F) in target.json file. Serial Output is not printing anything.
fpu is supported by my controller.

-----------------------------------------------------------------

## Question
How to resolve this issue?

amq, 2017-07-18:
@pranavshandilya what is the MCU? I don't think it is possible to help you without knowing at least the family, e.g. [STM32F437xG](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F437xG) or [EFM32PG12](https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_Silicon_Labs/TARGET_EFM32/TARGET_EFR32MG12).

What is the mbed-os version?

Does it crash on printf (add a Ticker-based blinky to check this), or does it simply not print anything?

0xc0170, 2017-07-18:
As above , more details. Have you verified that data are not coming out from TX? does RX work? Does an application crash somwhere? is it running and sending but no data are going out of TX line? What toolchains have you tested it? However I would guess this is hw related problem rather than toolchain library.

There are same core targets in the codebase , no problems with serial output.

pranavshandilya, 2017-07-18:
@amq @0xc0170 thanks for ur quick reply!

>What is the mbed-os version?
MbedOS 5.5.2

That is the custom board, which is not listed in target.json file(MCU: TM4C129X Series).

It's working fine with Cortex-M4 but when I changed to Cortex-M4F it's not printing anything.

When I export to keil(using Cortex-M4F) for debugging, it's giving Hardfault handler while using Cortex-M4 it's working fine as I expected.

@0xc0170 
I'm tested with ARM.

> However I would guess this is hw related problem rather than toolchain library.
If it's hw related issue then how is it working with Cortex-M4?

amq, 2017-07-18:
I'd say the issue is invalid without the official MCU family support, or at the very least without you making the target files public.

pranavshandilya, 2017-07-18:
@amq 

I'll try to public the driver code ASAP...!
Thanks for ur support!

closed 2017-07-18 by pranavshandilya