ARMmbed/mbed-os #4773: Ethernet interface connect fails in Release on Nucleo STM32F

avolinski, 2017-07-18:
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NUCLEO STM32F

**Toolchain:**
GCC_ARM, ARM , IAR 
On all 3 toolchains happens is **Release** Only

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**mbed-os sha:**
latest tag

**DAPLink version:**

**Expected behavior**
Ethernet connect succeeds 

**Actual behavior**
connect seems to be stuck forever

**Steps to reproduce**

EthernetInterface* netInterface = NULL;
netInterface = new EthernetInterface();
netInterface->connect();

betzw, 2017-07-19:
PR #4768 might resolve this issue.
 cc @bcostm @screamerbg @0xc0170

0xc0170, 2017-07-24:
@avolinski Have you had a chance to try the PR referenced above if it fixes this issue?

jeromecoutant, 2017-08-17:
Hi @avolinski 

Could you precise which target do you use ?

From my side, I got some issue with STM32F7 only.
I have check #4808 and #4809 but there are still some IP connect issue

Regards,

JanneKiiskila, 2017-08-18:
I believe we're using NUCLEO F429_ZI.

avolinski, 2017-08-20:
Issue still exists with mbedOS 5.5.5 release

SenRamakri, 2017-08-23:
We have confirmed that, at least one of the Ethernet connectivity issue has been resolved by below
PR.

https://github.com/ARMmbed/mbed-os/pull/4808

Please note that this fix is not in 5.5.5 Release, should get in 5.5.6 release. If you want to verify with this fix, use mbed-os "latest" which has this fix merged.

0xc0170, 2017-09-04:
@avolinski Can you confirmed its fixed on master or update this issue?

teetak01, 2017-09-04:
This has been fixed in mbed OS 5.5.6

closed 2017-09-04 by 0xc0170