ARMmbed/mbed-os #4785: Since mbed-os 5.5.0 STM32 Nucleo L476RG - Serial looses bytes with communication with uCam-III

sergiuszm, 2017-07-20:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
STM32 NUCLEO L476RG

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3 20150529

**mbed-cli version:** 
1.1.1

**mbed-os sha:**
>= 5.5.0

**Expected behavior**
Defined callback for Serial::RxIrq should receive 512 bytes packages from external device.

**Actual behavior**
Defined callback for Serial::RxIrq randomly received less than 512 bytes packages from external device.

The output below illustrates how many bytes was received by a callback function. Data size should be increased by 512 bytes except the last package. The numbers after -> shows the result of my data buffer with module 512, so I can show you at least few packages in which dew bytes were lost. I have tried to using baud rated from 57600 up to 1228800 without success.
```
Image size: 24276
Number of packages: 48
Received data size: 512 -> 0
Received data size: 1024 -> 0
Received data size: 1536 -> 0
Received data size: 2048 -> 0
Received data size: 2559 -> 511
Received data size: 3071 -> 511
Received data size: 3583 -> 511
Received data size: 4095 -> 511
Received data size: 4607 -> 511
Received data size: 5119 -> 511
Received data size: 5631 -> 511
Received data size: 6143 -> 511
Received data size: 6655 -> 511
Received data size: 7166 -> 510
Received data size: 7678 -> 510
Received data size: 8189 -> 509
Received data size: 8701 -> 509
Received data size: 9213 -> 509
Received data size: 9725 -> 509
Received data size: 10237 -> 509
Received data size: 10749 -> 509
Received data size: 11261 -> 509
Received data size: 11773 -> 509
Received data size: 12285 -> 509
Received data size: 12796 -> 508
Received data size: 13308 -> 508
Received data size: 13819 -> 507
Received data size: 14331 -> 507
Received data size: 14843 -> 507
Received data size: 15355 -> 507
Received data size: 15867 -> 507
Received data size: 16379 -> 507
Received data size: 16891 -> 507
Received data size: 17403 -> 507
Received data size: 17915 -> 507
Received data size: 18427 -> 507
Received data size: 18939 -> 507
Received data size: 19450 -> 506
Received data size: 19962 -> 506
Received data size: 20473 -> 505
Received data size: 20985 -> 505
Received data size: 21497 -> 505
Received data size: 22009 -> 505
Received data size: 22521 -> 505
Received data size: 23033 -> 505
Received data size: 23545 -> 505
Received data size: 24057 -> 505
Received data size: 24557 -> 493

Callback function itself:
```
void uCam::_rx_callback() {
    _image_buffer[_img_buffer_pos++] = _uCam->getc();
}
```

**Steps to reproduce**
I am not pretty sure how it can be reproduced without uCam-III. The thing worth notice is, that BufferedSerial is used on the top of the Serial class in order to buffer received data. BufferedSerial from [ARMmbed/ATParser](https://github.com/ARMmbed/ATParser/tree/master/BufferedSerial) is used.

The same code works well with mbed-os 5.4.x up to 5.5.0 when the problem started to occur.

The output below illustrates the correct number of bytes received by the callback function, that works with mbed-os 5.4.x
```
Image size: 24188
Number of packages: 48
Received data size: 512 -> 0
Received data size: 1024 -> 0
Received data size: 1536 -> 0
Received data size: 2048 -> 0
Received data size: 2560 -> 0
Received data size: 3072 -> 0
Received data size: 3584 -> 0
Received data size: 4096 -> 0
Received data size: 4608 -> 0
Received data size: 5120 -> 0
Received data size: 5632 -> 0
Received data size: 6144 -> 0
Received data size: 6656 -> 0
Received data size: 7168 -> 0
Received data size: 7680 -> 0
Received data size: 8192 -> 0
Received data size: 8704 -> 0
Received data size: 9216 -> 0
Received data size: 9728 -> 0
Received data size: 10240 -> 0
Received data size: 10752 -> 0
Received data size: 11264 -> 0
Received data size: 11776 -> 0
Received data size: 12288 -> 0
Received data size: 12800 -> 0
Received data size: 13312 -> 0
Received data size: 13824 -> 0
Received data size: 14336 -> 0
Received data size: 14848 -> 0
Received data size: 15360 -> 0
Received data size: 15872 -> 0
Received data size: 16384 -> 0
Received data size: 16896 -> 0
Received data size: 17408 -> 0
Received data size: 17920 -> 0
Received data size: 18432 -> 0
Received data size: 18944 -> 0
Received data size: 19456 -> 0
Received data size: 19968 -> 0
Received data size: 20480 -> 0
Received data size: 20992 -> 0
Received data size: 21504 -> 0
Received data size: 22016 -> 0
Received data size: 22528 -> 0
Received data size: 23040 -> 0
Received data size: 23552 -> 0
Received data size: 24064 -> 0
Received data size: 24476 -> 412
```
## Question
If you have any ideas how I can help with the issue debugging feel free to let me know. If you have any other questions let me know as well.

Data receiving process looks like a hand-shake. At the beginning I am sending request for a first package of 512 bytes. Then after receiving package my code is sending ACK to the external device and in return receives next package of 512 bytes until the end of data in external device memory.

0xc0170, 2017-07-20:
cc @bcostm @adustm @LMESTM @jeromecoutant

bulislaw, 2017-07-20:
Issues with idle sleep? Could you try chaning that:
```
diff --git a/targets/targets.json b/targets/targets.json
index ebad04652..ea7917c9a 100644
--- a/targets/targets.json
+++ b/targets/targets.json
@@ -674,7 +674,7 @@
         "extra_labels": ["STM"],
         "supported_toolchains": ["ARM", "uARM", "IAR", "GCC_ARM"],
         "macros": ["TRANSACTION_QUEUE_SIZE_SPI=2"],
-        "device_has": ["ANALOGIN", "I2C", "I2CSLAVE", "I2C_ASYNCH", "INTERRUPTIN", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SERIAL", "SLEEP", "SPI", "SPISLAVE", "SPI_ASYNCH", "STDIO_MESSAGES"]
+        "device_has": ["ANALOGIN", "I2C", "I2CSLAVE", "I2C_ASYNCH", "INTERRUPTIN", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SERIAL", "SPI", "SPISLAVE", "SPI_ASYNCH", "STDIO_MESSAGES"]
     },
     "NUCLEO_F030R8": {
         "inherits": ["FAMILY_STM32"],
```
and rerun your test.

sergiuszm, 2017-07-20:
@bulislaw Unfortunately, the same result.

sergiuszm, 2017-07-24:
It looks like everything is working even with > mbed-os 5.5.x right now. I believe I have had some kind of malfunctioning code, so I am closing this issue.

I'm sorry for the false alarm.

closed 2017-07-24 by sergiuszm