ARMmbed/mbed-os #4787: Unable to establish BLE connexion on NRF52840_DK target

lu0m4n, 2017-07-20:
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
NRF52840_DK

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
a67c729 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #97 from 0xc0170/master

**DAPLink version:**

**Expected behavior**
When the device is advertising, we should be able to establish a BLE connexion.

**Actual behavior**
The connexion fails with a timeout error.

**Steps to reproduce**
Using a NRF52840_DK target, build any of the mbed BLE example and run it.

0xc0170, 2017-07-20:
cc @pan- @nvlsianpu

pan-, 2017-07-20:
@lu0m4n I've just tried with the example `BLE_BatteryLevel` and I have no trouble connecting to the device.

lu0m4n, 2017-07-21:
@pan Thanks for the feedback. Could you please share details about your environment, mbed version, board revision ... ?

pan-, 2017-07-21:
@lu0m4n 

**Target**
NRF52840_DK (10056)

**Toolchain:**
GCC_ARM: arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]

**mbed-cli version:** 
1.2.0

**mbed-os-ble-examples sha:**
a67c729

lu0m4n, 2017-07-27:
Thank you @pan- 

I swtiched from LightBlue to nRFConnect as debug application, and noticed in logs that the connection is establishing succesfully. So I don't know why it doesn't manage to display GATT database services and characteristics.

lu0m4n, 2017-07-28:
I tried on an Android tablet and it works well. It appears the problem has nothing to do with mbed but is specific to iOS device. Anyway, thank you for your help.

closed 2017-07-28 by lu0m4n

pan-, 2017-07-28:
@lu0m4n Is the discovery of services automatic in LightBlue ? That may be your issue.