ARMmbed/mbed-os #4788: Can't Export

alphatronicstva, 2017-07-20:
### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO-F429ZI

**Actual behavior**
When i press export in the online compiler/in an example project it gives an error.

**Steps to reproduce**
Try to export a project to system workbench with target NUCLEO-F429ZI
----------------------------------------------------------------

0xc0170, 2017-07-20:
cc @theotherjimmy

alphatronicstva, 2017-07-21:
I have pictures if i can upload them somewhere?

marcuschangarm, 2017-07-21:
You can drag-and-drop pictures right here in the comment box.

theotherjimmy, 2017-07-21:
System workbench is not an exporter, as far as I know. Which exporter could be called system workbench?

theotherjimmy, 2017-07-21:
You do not need to upload the compiler's "export failed" dialog. I have seen it plenty :)

JojoS62, 2017-07-21:
> Which exporter could be called system workbench?
SW4STM32 -> system workbench for STM32 :)

theotherjimmy, 2017-07-21:
Oh, Thanks!

theotherjimmy, 2017-07-24:
@alphatronicstva You should be able to work around this problem the following steps. I have added the command line instructions where appropriate. Please fill in the sections between the `<` and `>` by replacing the entire word, including the `<` and `>`. For example if you project were named `mbed-os-example-blinky` the second command in step 1 would be `cd mbed-os-example-blinky`.

 1) Pull down your project with mbed CLI: `mbed import <you-project-repo-here>` then `cd <your-project-name-here>`
 2) Export the project 
    1) For a managed project: `mbed export -i gnuarmeclipse -m <your-dev-board-here>` 
    2) For an unmanaged project: `mbed export -i eclipse_gcc_arm -m <your-dev-board-here>` 
 3) Copy the exact location of the project for use later. `pwd` then select and copy
 4) Import the project into the IDE. 
    1) Select `import` from the `file` menu
    2) Select `Existing Projects into Workspace` from the `General` section
    3) Paste the directory that you copied in step 3 into the "Select root directory" section.
    4) Click finish.

alphatronicstva, 2017-07-24:
@theotherjimmy thank you, how i know my repo name? https://developer.mbed.org/compiler/#nav:/my_project; is the place I work on it?

theotherjimmy, 2017-07-24:
@alphatronicstva You can make your repo name from the developer site by:

1) Click on the "pubilsh" button in the online compiler.
2) Fill out the forms (select "private" if you don't want anyone else to be able to see your repo)
3) Finally, a dialog will pop up telling you the repo of your project.

theotherjimmy, 2017-07-25:
@alphatronicstva Were you able to get things working? Would you like anymore help?

alphatronicstva, 2017-07-26:
@theotherjimmy i have problems with authentication, when i enter the correct username and password it says authorization failed ...

theotherjimmy, 2017-07-26:
During the `mbed import` I take it. You could try adding the `-v` flag, or using hg yourself. Doing `mbed import -v <your-repo` will give you the commands it's running, so that can be a good way to debug things here. 

@immunda Do we know of any issues that @alphatronicstva might be facing with login through hg?

I agree that this is getting hairy, and I think it's all a bunch of small problems. In the mean time, We got a new version of the website deployed that should have resolved a large number of the exporter errors. I just tried exporting a project in my workspace to SW4STM32 for the Nucleo-F429ZI, and I got a zip file. @alphatronicstva Give that a try.

alphatronicstva, 2017-07-27:
@theotherjimmy i've tried to export it again but the same error. I'll try to do it in cmd line with the v flag on.

alphatronicstva, 2017-07-27:
The problem with the mbed import was that i've used my git account criteria, not my mbed account criteria. Now it works.

alphatronicstva, 2017-07-27:
[image]
Now i have this problem.

theotherjimmy, 2017-07-27:
Looking at https://developer.mbed.org/teams/ST/code/Nucleo_blink_led/shortlog I have to agree with mbed CLI on this one: I don't see that revision either.

Well, you could try to find what the revision is called, and update the `.lib` file with that revision. But really this looks like a problem with the project you're importing, and the fact that it compiles in the online compile means that I'm missing a piece of the puzzle here. Are you adding the "Nucleo_blink_led" project as a `.lib` file to your project?

alphatronicstva, 2017-07-27:
@theotherjimmy it has nothing to do with blinking led, i chose that template when creating a project.

theotherjimmy, 2017-07-27:
Thanks for the clarification.

We have made progress: your repo should now be on your local machine. Now we have to figure out what went wrong with that library checkout. Is the project at the right revision now? that is: have the changes that you made in the online compiler made their way to your local machine?

theotherjimmy, 2017-07-27:
Another question that is important, now that I know what your project is based on: When you clicked publish, did you fork the repo into your user acount? If not, please do that. You should notice that the link to your repo has changed after you fork. Try `mbed import`ing that url instead.

theotherjimmy, 2017-07-27:
Further, If you could provide me with your developer.mbed.org username, I could check the logs for you to see if I can find what went wrong with your online export.

alphatronicstva, 2017-07-27:
@theotherjimmy not everything is on my machine. But i'll retry. And yes i've forked it private to my account.

theotherjimmy, 2017-07-27:
@alphatronicstva Then the url in the IDE should be correct for the root project. From your screenshot it looked like you were cloning the ST Nucleo_Blink_Led project. That's why I thought the URL might be wrong.

Another possibility: are you depending on libraries that you have modified? If so, you will have to publish  and fork those as well, so that the URL is importable by mbed CLI.

alphatronicstva, 2017-07-27:
It doesn't export everything ... 
[image]

alphatronicstva, 2017-07-27:
This is my buildup for the project.

[image]

alphatronicstva, 2017-07-27:
@theotherjimmy all edited stuff is forked. I can't start from a empty project in the online compiler so...

theotherjimmy, 2017-07-27:
@alphatronicstva Thanks for the screenshots. They provide some helpful information. 

Thanks for the clarification on the forking. I think we are narrowing down the error here.

alphatronicstva, 2017-07-27:
[image]

I can download a zip with the main.cpp and lib files but i can't export (see link).

theotherjimmy, 2017-07-27:
It looks like something went wrong with the clone, as it's looking for the "Nucleo_blink_led" project to add as a library. That's clearly incorrect. If you have a local clone of the repo, you could check that you only have `.lib` files for the dependent libraries, and a `.bld` for mbed. Any other `.lib` files should be deleted.

From the root of your project, you should have these files:
 * main.cpp
 * LIS3DH.lib
 * PN532.lib
 * ST7567.lib
 * mbed.bld

And that's it. If anything else is there, delete it. Then run `mbed deploy` from the root to get all of your dependencies.

theotherjimmy, 2017-07-27:
I have another idea: when it complained about the folder not being empty, did you remove the entire folder, or delete its contents? If it's the latter, try deleting the folder and just starting anew.

alphatronicstva, 2017-07-27:
[image]

theotherjimmy, 2017-07-27:
@alphatronicstva mbed CLI should not be doing that unless you have a `.lib` file named `F429_LCD_ST7567.lib` in your project with the contents `https://developer.mbed.org/teams/ST/code/Nucleo_blink_led#49ac5db20f5a`. Can you check for something such as that?

alphatronicstva, 2017-07-28:
@theotherjimmy I really don't understand what you mean, but when i do the project again from 0 what to do? I'm just going to copy paste everything in a new project.

theotherjimmy, 2017-07-28:
Copy paste these files only:
 * main.cpp
 * LIS3DH.lib
 * PN532.lib
 * ST7567.lib
 * mbed.bld

Then you should be good to go.

alphatronicstva, 2017-08-01:
Since today the export button works from the online compiler. But i have another problem now.
When I try to do MBED deploy it does this?

[image]

alphatronicstva, 2017-08-01:
Rev history:

[image]

theotherjimmy, 2017-08-01:
@alphatronicstva You got to the bottom of it! That highlighted lib file should not be there.

theotherjimmy, 2017-08-01:
If you just remove the file locally, you should be able to deploy.

alphatronicstva, 2017-08-01:
@theotherjimmy i figured that out myself this afternoon. But now it gives alot of compiler errors, so i need to change the code. Thanks for the good support!!!

theotherjimmy, 2017-08-01:
@alphatronicstva You're welcome. Just ping me if you want help anytime. You mentioned that this issue was resolved a few comments above, so I'm closing this issue.

closed 2017-08-01 by theotherjimmy