ARMmbed/mbed-os #4789: Spi endianness

alphatronicstva, 2017-07-20:
We need to be able to say to each SPI object if it is LSB first or MBS first. This isn't implemented at the moment.

0xc0170, 2017-07-20:
cc @bulislaw @c1728p9

bulislaw, 2017-07-21:
We need more info including detailed description, context and examples. Can you provide that please.

alphatronicstva, 2017-07-21:
Dear

What do you mean, I need to set the endianness for each spi object. Example: spi1 is LSB & spi2 is MSB. This is supported normally by ST or other chip producers. We use the STM32F429 for this test.

Greetings
Thomas Van Aken

bulislaw, 2017-07-21:
@sg- @c1728p9 is there a reason our HAL doesn't support setting this parameters for targets SPI implementations?

0xc0170, 2017-07-21:
> @sg- @c1728p9 is there a reason our HAL doesn't support setting this parameters for targets SPI implementations?

I do not know any. I have seen it and was asked for it previously. The most common is MSB that is by default for SPI. Users usually reverse bits prior the transfer (software emulation). I recall efm32, st, nxp chips support this settings.

@alphatronicstva Out of curiosity, what module requires LSB first?

alphatronicstva, 2017-07-21:
PN532, but SW emulation is too much overhead, we tought about this, but we need to do this every time we receive and transmitt a byte so thats not ideal. I prefer the hardware solution.

alphatronicstva, 2017-07-23:
Is there a fix possible? We have 2 SPI objects and for our screen it needs to be MSB first, for our RFID library that I've proted from eclipse to mbed it needs to be LSB first. Our project is on a tight timeline, i hope there is a solution soon. Editing the library to solve this problem is not ideal at al.

bulislaw, 2017-07-24:
We are in progress (beginning) of reworking our HAL layer we will consider adding it to the API when we get around to SPI. But that's not going to be very soon, I can't really give you any estimate.

deepikabhavnani, 2018-09-11:
SPI bit ordering is consider in new SPI HAL specifications https://github.com/ARMmbed/mbed-os/pull/7671/files#diff-0b914e1ff4687e634d1563b826f773a9R143

Can be closed

closed 2018-09-11 by deepikabhavnani