ARMmbed/mbed-os #4791: Compile failing when project path contains non-ascii character

xymopen, 2017-07-20:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**platform:** 
(`Windows 10 64 Bit`)

**Python version:** 
(`2.7.13 64 Bit`)

**mbed-cli version:** 
(`1.2.0`)

**mbed-os sha:**
(`692d905`)

**Steps to reproduce**
    # 示例 means example
    mkdir 示例
    cd 示例
    mbed import mbed-os
    cd mbed-os
    mbed compile -m DISCO_F407VG -t GCC_ARM

**Actual behavior**
    Building project mbed-os (DISCO_F407VG, GCC_ARM)
    Scan: .
    Scan: env
    Scan: mbed
    Compile [  0.3%]: AnalogIn.cpp
    Compile [  0.7%]: BusInOut.cpp
    Compile [  1.0%]: BusIn.cpp
    Compile [  1.3%]: BusOut.cpp
    Compile [  1.6%]: FlashIAP.cpp
    Compile [  2.0%]: CAN.cpp
    Compile [  2.3%]: Ethernet.cpp
    Compile [  2.6%]: I2CSlave.cpp
    Compile [  2.9%]: I2C.cpp
    Compile [  3.3%]: InterruptIn.cpp
    Compile [  3.6%]: InterruptManager.cpp
    Compile [  3.9%]: RawSerial.cpp
    Compile [  4.2%]: SPISlave.cpp
    Compile [  4.6%]: SPI.cpp
    Compile [  4.9%]: Serial.cpp
    Compile [  5.2%]: SerialBase.cpp
    Compile [  5.5%]: Ticker.cpp
    Compile [  5.9%]: Timeout.cpp
    Compile [  6.2%]: TimerEvent.cpp
    Compile [  6.5%]: Timer.cpp
    Compile [  6.8%]: UARTSerial.cpp
    Compile [  7.2%]: equeue_posix.c
    Compile [  7.5%]: EventQueue.cpp
    Compile [  7.8%]: equeue_mbed.cpp
    Compile [  8.1%]: equeue.c
    Compile [  8.5%]: Dir.cpp
    Compile [  8.8%]: File.cpp
    Compile [  9.1%]: FileSystem.cpp
    Compile [  9.4%]: ChainingBlockDevice.cpp
    Compile [  9.8%]: ccsbcs.cpp
    Compile [ 10.1%]: HeapBlockDevice.cpp
    Compile [ 10.4%]: SlicingBlockDevice.cpp
    Compile [ 10.7%]: MBRBlockDevice.cpp
    Compile [ 11.1%]: ff.cpp
    Compile [ 11.4%]: greentea_metrics.cpp
    Compile [ 11.7%]: greentea_serial.cpp
    Compile [ 12.1%]: FATFileSystem.cpp
    Compile [ 12.4%]: greentea_test_env.cpp
    Compile [ 12.7%]: unity.c
[... 250 lines cut ...]
    Compile [ 93.2%]: spi_api.c
    Compile [ 93.5%]: serial_device.c
    [Warning] serial_device.c@290,35: unused variable 'tmpval' [-Wunused-variable]
    Compile [ 93.8%]: analogout_api.c
    Compile [ 94.1%]: can_api.c
    Compile [ 94.5%]: gpio_api.c
    Compile [ 94.8%]: hal_tick_16b.c
    Compile [ 95.1%]: gpio_irq_api.c
    Compile [ 95.4%]: hal_tick_32b.c
    Compile [ 95.8%]: lp_ticker.c
    Compile [ 96.1%]: mbed_overrides.c
    Compile [ 96.4%]: pinmap.c
    Compile [ 96.7%]: i2c_api.c
    Compile [ 97.1%]: port_api.c
    Compile [ 97.4%]: pwmout_api.c
    Compile [ 97.7%]: rtc_api.c
    Compile [ 98.0%]: serial_api.c
    Compile [ 98.4%]: trng_api.c
    Compile [ 98.7%]: sleep.c
    [ERROR] 'ascii' codec can't decode byte 0xcf in position 0: ordinal not in range(128)
    [mbed] ERROR: "c:\program files\python\2.x\python.exe" returned error code 1.
    [mbed] ERROR: Command "c:\program files\python\2.x\python.exe -u E:\项目\示例\mbed-os\tools\make.py -t GCC_ARM -m DISCO_F407VG --source . --build .\BUILD\DISCO_F407VG\GCC_ARM" in "E:\项目\示例\mbed-os"
    ---

Removing the non-ascii characters in project path resolves the issue.

geky, 2017-07-20:
cc @theotherjimmy

Nodraak, 2017-07-21:
Is a Pyhton 3 migration planned? I believe that it would solve a lot of utf-8-related issues.

theotherjimmy, 2017-07-21:
@Nodraak Yes it would solve this immediate problem. We have a few dependencies on other packages that are python 2 only, so we're waiting on them. Further, if we upgrade to python 3, we would have to update mbed CLI to call python 3, which would break old versions of the tools...

Nodraak, 2017-07-21:
I understand. Is there an issue or something that keeps track of which packages are not supported by py3 yet? Concerning mbed cli, maybe it would be feasable to support both python version, or otherwise find a way for mbed cli to detect which python version should be used (based on the shebang? It's a bit hacky, but could be a solution).

theotherjimmy, 2017-07-21:
> based on the shebang? It's a bit hacky, but could be a solution

I was going to suggest that. and default to python 2 when no shebang is present should be backwards compatible.

theotherjimmy, 2017-07-21:
Anyway. @xymopen I'm going to get this working for you soon. I'll mention you on the PR when I put it up.

xymopen, 2017-07-22:
@theotherjimmy Thanks.

cmonr, 2018-06-25:
Interesting. Definitely looks like an issue. Will start looking into how to solve this, with some new tests to come.

closed 2021-10-20 by ciarmcom