ARMmbed/mbed-os #4798: Lwm2m blockwise data transfer (using Block1 option) with Leshan not working

ghseb, 2017-07-24:
There possibly is bug in sn_coap_protocol.c that breaks Block1 data transfer from Leshan servers to mbed clients.

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**mbed-os sha:**
8828635

**Actual behavior**
When transmitting data as block-wise PUT (Coap Block1) from Leshan to a mbed device the token of the acknowledgement (Coap Continue) differs from token of the block-message. Leshan then thinks that the block transfer has failed and no further blocks are sent.

**Expected behavior**
Coap Continue uses the same token as the Block1 Message it acknowledges.

**Suggested solution**
Copy the token of the Block1 message into the acknowledgement (Coap Continue).

0xc0170, 2017-07-24:
cc @SeppoTakalo

SeppoTakalo, 2017-07-24:
CC: @yogpan01 @JanneKiiskila This CoAP library is owned by client teams. Please check this bug.

JanneKiiskila, 2017-07-24:
@anttiylitokola - can you check this?

anttiylitokola, 2017-07-26:
@JanneKiiskila, Fix is waiting for review.
https://github.com/ARMmbed/mbed-coap/pull/25

closed 2017-08-24 by ghseb