ARMmbed/mbed-os #4802: Exported files in subfolders are copied to project root

johanstokking, 2017-07-24:
### Description
- Type: Bug
- Introduced in: #4544 
- Priority: Minor

---------------------------------------------------------------
## Bug

**mbed-cli version:** 
`1.2.0`

**mbed-os sha:**
Since #4544 got merged, current master for example: `d5108e5a7`

**Expected behavior**
There should be no files copied to project root that are exported to a subfolder already.

**Actual behavior**
When exporting with `mbed-cli` with an exporter that exports in subfolders the files are copied to project root. For example, [`VSCode` exports `.vscode/{tasks|launch|settings}.json`](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/vscode/__init__.py#L43-L48).

**Steps to reproduce**
```
$ git clone https://github.com/ARMmbed/mbed-os-example-blinky.git
$ cd mbed-os-example-blinky
$ mbed deploy
$ mbed export -i vscode_gcc_arm -m NRF52_DK --profile mbed-os/tools/profiles/debug.json
$ find . | grep -E '(launch|settings|tasks).json'
./.vscode/launch.json
./.vscode/settings.json
./.vscode/tasks.json
./launch.json
./settings.json
./tasks.json
```

Those last three files shouldn't be there.

theotherjimmy, 2017-07-24:
@johanstokking Thanks for the bug report. I'm going to look into this.

theotherjimmy, 2017-07-24:
NICE! or you can do it!

PR #4804 (2017-07-27): Copy static files only

## Description
Closes #4802 
Replaces #4803 

## Status
**READY**
closed 2017-07-27 by theotherjimmy