ARMmbed/mbed-os #4806: Build error occurs by using branch that supports CMSIS5/RTX5 for Cortex-A

TomoYamanaka, 2017-07-25:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
RZ_A1H

**Toolchain:**
ARM

**Toolchain version:**
5.06u1

**mbed-cli version:** 
1.1.1

______
@bulislaw

I tried the below branch(feature_cortex_a) that is supporting CMSIS5/RTX5 for Cortex-A for mbed OS 5.6.
https://github.com/ARMmbed/mbed-os/tree/feature_cortex_a

The program I tried is `mbed-os-example-blinky`, the tool-chain is ARMCC.
I changed mbed-os library to the above branch.

As the result of the build, there were two problems for now.
**[1st problem]**
An error occurred that `mbed_rtx.h` is missing
It seems that there isn't for Cortex-A, there is for Cortex-M.
I brought this file from the past program(e.g. mbed OS 5.4.7) and avoided it.
Could you please revert this file?

**[2nd problem]**
There were the below error messages.
```
[Error] CThunk.h@225,0: #20: identifier "__v7_clean_inv_dcache_mva" is undefined
[Error] CThunk.h@228,0: #20: identifier "__v7_inv_icache_all" is undefined
[Error] CThunk.h@229,0: #20: identifier "__ca9u_inv_tlb_all" is undefined
[Error] CThunk.h@230,0: #20: identifier "__v7_inv_btac" is undefined
```
I consider that cause of those error is the lack of file that defined substance.
In the previous program(e.g. mbed OS 5.4.7), those functions have been implemented in `core_caFunc.h`.
(refer to https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.4/cmsis)
But in above branch, it seems that the substance of those functions did not existed because  `core_caFunc.h` has been removed by CMSIS5 update.

JonatanAntoni, 2017-07-25:
The `core_caFunc.h` is not part of CMSIS, please look for `core_ca.h` in CMSIS devel branch.

`__v7_clean_inv_dcache_mva` is called `L1C_CleanInvalidateDCacheMVA` 
`__v7_inv_icache_all` is called `L1C_InvalidateICacheAll`
`__v7_inv_btac` is called `L1C_InvalidateBTAC`
`__ca9u_inv_tlb_all` is called `MMU_InvalidateTLB`

bulislaw, 2017-07-25:
@TomoYamanaka I've updated RTX and fixed some of the issues on `feature_cortex_a` branch. Now I'm getting `mbed-os\rtos\mbed_boot.c", line 216: Error:  #35: #error directive: "no target defined"`

Which seems to be because the symbols are not exposed `HEAP_START` and/or `INITIAL_SP` in `startup_MBRZA1H.S`.

bulislaw, 2017-08-03:
@TomoYamanaka @toyowata ping, any news?

TomoYamanaka, 2017-08-03:
@bulislaw 

Sorry, there is no news for now.

> Which seems to be because the symbols are not exposed  HEAP_START  and/or  INITIAL_SP  in   startup_MBRZA1H.S .

Could you fix this issue?

toyowata, 2017-08-04:
@TomoYamanaka You may need to add proper INITIAL_SP definition as @bulislaw suggestion something like:
```
#define INITIAL_SP 0x00008000
```

You will probably see following errors, then.

```
Compile [ 99.3%]: us_ticker.c
(snip)
[Error] vfp_neon_push_pop.h@9,0:  #65: expected a ";"
[Error] vfp_neon_push_pop.h@29,0:  #10: "#" not expected here
[Error] vfp_neon_push_pop.h@55,0:  #10: "#" not expected here
(snip)
```

For this, you need to change ```__STATIC_ASM``` modifier to be ```__asm```.

```
__STATIC_ASM void __vfp_neon_push(void) {
__asm void __vfp_neon_push(void) {
```

For CDAbtHandler symbol multiple defined error, you need to remove these code from system_MBRZA1H.c because handers.c also has same definitions.

https://github.com/ARMmbed/mbed-os/blob/feature_cortex_a/rtos/rtx5/RTX/Config/TARGET_CORTEX_A/handlers.c#L52

TomoYamanaka, 2017-08-08:
@toyowata 

Thank you for your comments. And sorry for this late relpy.
I will try it soon.

TomoYamanaka, 2017-08-10:
@bulislaw @toyowata 

I changed the below contents, however it seems that build errors still occur.
* Add `INITIAL_SP` macro in `mbed_rtx.h`
```
#ifndef INITIAL_SP
#define INITIAL_SP              (0x20020000UL)
#endif
```

* Change function type from` __STATIC_ASM` to `static __asm` in `vfp_neon_push_pop.h`

* Remove the below funtions in `system_MBRZA1H.c`
```
void CDAbtHandler(uint32_t DFSR, uint32_t DFAR, uint32_t LR) 
void CPAbtHandler(uint32_t IFSR, uint32_t IFAR, uint32_t LR)
uint32_t CUndefHandler(uint32_t opcode, uint32_t state, uint32_t LR)
```

* Change function type to `__WEAK` in `rtx_lib.c`
`__WEAK int _mutex_initialize(mutex *m) `

* Add the below definition in `target.json`
"macros": `["__GIC_PRESENT=1", "GIC_DISTRIBUTOR_BASE=0xe8201000UL", "GIC_INTERFACE_BASE=0xe8202000UL", "TIMER_BASE=0xFCFEC000", "__TIM_PRESENT=1", "__CORTEX_A=9"], `

* Remove `gic.c` and `gic.h` for re-definition

Currently, I face the below build error.
```
[Error] core_ca.h@1362,0:  #511: this operation on an enumerated type requires an applicable user-defined operator function
[Error] core_ca.h@1399,0:  #511: this operation on an enumerated type requires an applicable user-defined operator function
```

toyowata, 2017-08-11:
@TomoYamanaka 

>Currently, I face the below build error.
>
>```
>[Error] core_ca.h@1362,0:  #511: this operation on an enumerated type requires an applicable user-defined operator function
>[Error] core_ca.h@1399,0:  #511: this operation on an enumerated type requires an applicable user-defined operator function
>```

I think [this](https://github.com/ARMmbed/mbed-os/blob/feature_cortex_a/cmsis/TARGET_CORTEX_A/core_ca.h#L1361
) needs to have a cast to avoid the error.

```cpp
for (i = (IRQn_Type)32; i < num_irq; (IRQn_Type)i++)
```

bulislaw, 2017-08-14:
@JonatanAntoni what do you think?

TomoYamanaka, 2017-08-15:
@toyowata 

> I think this needs to have a cast to avoid the error.

It's seems that I had missed it.
Thanks for your comment.

toyowata, 2017-08-23:
@TomoYamanaka 

It seems to have more fixes to build Cortex-A9 target in the feature branch.

# Renesas target code

* Following symbol in the mmu_Renesas_RZ_A1.c has to be replaced new function names as below:

Previous calling function names|It should be replaced to...
------------ | -------------
|__TTSection|MMU_TTSection|

* Following symbols in the system_MBRZA1H.c have to be replaced new function names as below:

Previous calling function names|It should be replaced to...
------------ | -------------
|__ca9u_inv_tlb_all|MMU_InvalidateTLB|
|__enable_btac|L1C_EnableBTAC|
|__enable_caches|L1C_EnableCaches|
|__enable_mmu|MMU_Enable|
|__v7_inv_btac|L1C_InvalidateBTAC|
|__v7_inv_dcache_all|L1C_InvalidateDCacheAll|
|__v7_inv_icache_all|L1C_InvalidateICacheAll|

# Common code for Cortex-A

* Macro ```section_normal_nc``` needs to be added in the [core_ca.h](https://github.com/ARMmbed/mbed-os/blob/feature_cortex_a/cmsis/TARGET_CORTEX_A/core_ca.h#L1753).

```c
#define section_normal_nc(descriptor_l1, region)     region.rg_t = SECTION; \
                                   region.domain = 0x0; \
                                   region.e_t = ECC_DISABLED; \
                                   region.g_t = GLOBAL; \
                                   region.inner_norm_t = NON_CACHEABLE; \
                                   region.outer_norm_t = NON_CACHEABLE; \
                                   region.mem_t = NORMAL; \
                                   region.sec_t = SECURE; \
                                   region.xn_t = EXECUTE; \
                                   region.priv_t = RW; \
                                   region.user_t = RW; \
                                   region.sh_t = NON_SHARED; \
                                   MMU_GetSectionDescriptor(&descriptor_l1, region);
```

* ```#if defined(PTIM)``` position [here ](https://github.com/ARMmbed/mbed-os/blob/feature_cortex_a/rtos/rtx5/Source/os_tick_ptim.c#L25)should be after header file inclusion, not top of source file in os_tick_ptim.c..

Ref : https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/Source/os_tick_ptim.c#L31

toyowata, 2017-08-23:
@bulislaw @JonatanAntoni 

I quick checked GCC_ARM tool build in the feature_cortex_a branch and the fix [here](https://github.com/ARMmbed/mbed-os/issues/4806#issuecomment-321762854) is not proper solution since I got errors below:

```
[Error] core_ca.h@1361,41: no 'operator++(int)' declared for postfix '++' [-fpermissive]
[Error] core_ca.h@1398,46: no 'operator++(int)' declared for postfix '++' [-fpermissive]
```

I temporary changed the code below and both ARM and GCC_ARM can build without errors:

```c
__STATIC_INLINE void GIC_DistInit(void)
{
  uint32_t i;

  // snip

  for (i = 32; i < num_irq; i++)
  {
      //Disable the SPI interrupt
      GIC_DisableIRQ((IRQn_Type)i);
      if (i > 15) {
        //Set level-sensitive (and N-N model)
        GIC_SetConfiguration((IRQn_Type)i, 0);
      }
      //Set priority
      GIC_SetPriority((IRQn_Type)i, priority_field/2);
      //Set target list to CPU0
      GIC_SetTarget((IRQn_Type)i, 1);
  }
  //Enable distributor
  GIC_EnableDistributor();
}
```

Ref: https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core_A/Include/core_ca.h#L1363

bulislaw, 2017-08-23:
@RobertRostohar

TomoYamanaka, 2017-08-30:
I modified the value of `INITIAL_SP` as the below.
[old]
```
#define INITIAL_SP              (0x20020000UL)
```
[new]
```
#define INITIAL_SP              (0x20090000UL)
```

TomoYamanaka, 2017-08-30:
@bulislaw @toyowata

Current status is that the build was successed but not working yet.

Currently, we are debugging why it does not work,
as one of cause, we realized that there was a problem with the setting method of the heap.
The setting method of heap was different between Cortex-A and Cortex-M so far,
it seems that it became common processing by supporting CMSIS5 / RTX5.
Therefore, we are working the matching of code with reference to the processing of Cortex-M side.

bulislaw, 2017-08-30:
Great, thanks for the update.

TomoYamanaka, 2017-09-05:
@bulislaw @toyowata

**[ STATUS ]**
Current status is here.
- **ARMCC**
The build was successed, Blinky sample was worked.
However, we are facing the error at two test cases within the test used mbed CLI(mbed test ...).
```
tests-mbedmicro-rtos-mbed-basic
tests-mbedmicro-rtos-mbed-malloc
```
- **GCC_ARM**
The build was successed, Blinky sample was worked.
However, we are facing the error at three test cases within the test used mbed CLI.
```
tests-mbedmicro-rtos-mbed-basic
tests-mbedmicro-rtos-mbed-malloc
tests-mbed_drivers-timeout
```

<br><br>

**[ REQUEST ]**
It seems that it need to modify at the several main() functions of test. 

**1.**
[TEST\heap_and_stack](https://github.com/ARMmbed/mbed-os/blob/feature_cortex_a/TESTS/mbedmicro-rtos-mbed/memory/heap_and_stack/main.cpp
)
Previously, this test is not support in the test for Cortex-A, but this time it was implemented as the common test. However, we faced the error because [__get_MSP ()](https://github.com/ARMmbed/mbed-os/blob/feature_cortex_a/TESTS/mbedmicro-rtos-mbed/memory/heap_and_stack/main.cpp#L63) that used in the main function is a function not present in Cortex-A.
For this reason, for now, we put the following processing at the beginning of main.cpp and have skipped this test.
```
#if defined(TARGET_CORTEX_A)
　  #error [NOT_SUPPORTED] This function not supported for this target
#endif
```
If this test is also required for Cortex-A, would you please modify the processing in main() functoin ?

**2.**
[TEST\malloc](https://github.com/ARMmbed/mbed-os/blob/feature_cortex_a/TESTS/mbedmicro-rtos-mbed/malloc/main.cpp)
Although a macro [DEFAULT_STACK_SIZE](https://github.com/ARMmbed/mbed-os/blob/feature_cortex_a/TESTS/mbedmicro-rtos-mbed/malloc/main.cpp#L12) for Cortex-A used within the main function was defined in [cmsis_os.h](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.4/rtos/rtx/TARGET_CORTEX_A/cmsis_os.h#L73), it was gone by the support of CMSIS5 / RTX5.
For this reason, for now, we replaced the macro to `2048` by referencing the previous core_cafunc.h and excuted the test.
Would you please modify the processing in main() functoin ?

bulislaw, 2017-09-05:
> 1.
> TEST\heap_and_stack
> Previously, this test is not support in the test for Cortex-A, but this time it was implemented as the >common test. However, we faced the error because __get_MSP () that used [...]

That makes sense, I'll push this to the branch.

> 2.
> TEST\malloc
> Although a macro DEFAULT_STACK_SIZE for Cortex-A used within the main function was defined >in cmsis_os.h, it was gone by the support of CMSIS5 / RTX5.
> For this reason, for now, we [...]

We disallow changing main stack size for targets for compatibility sake. What exactly is the problem you are facing?

TomoYamanaka, 2017-09-06:
@bulislaw 

Thank you for your reply.

> 1.
> That makes sense, I'll push this to the branch.

I appreciate you.

> 2.
> We disallow changing main stack size for targets for compatibility sake. What exactly is the problem you are facing?

In the test used mbed CLI, we face the build error of referencing undefined macro(= `DEFAULT_STACK_SIZE`).
For example, the value of this macro was [2048](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.4/rtos/rtx/TARGET_CORTEX_A/cmsis_os.h#L73) in case of Cortex-A in mbed-os 5.4.
Therefore, we would appreciate that you modify `DEFAULT_STACK_SIZE` to `2048` in main() function of malloc test.

TomoYamanaka, 2017-09-06:
@bulislaw

Regarding `tests-mbed_drivers-timeout` of test case(toolcahin is GCC),
looking at the log(`mbed test ... -vv`), I understood it occurs timeout in the test. 
The log is here : [GCC(ver6)_Test_NG_tests-mbed_drivers-timeout.txt](https://github.com/ARMmbed/mbed-os/files/1280108/GCC.ver6._Test_NG_tests-mbed_drivers-timeout.txt)

However, in case of GCC version 4.9.3, this test was successed.
The log is here : [GCC(ver4.9.3)_Test_OK_tests-mbed_drivers-timeout.txt](https://github.com/ARMmbed/mbed-os/files/1280117/GCC.ver4.9.3._Test_OK_tests-mbed_drivers-timeout.txt)

<br>
I don't know why occurs timeout by defference of compiler version.
Could you give me any advice ?

TomoYamanaka, 2017-09-07:
@bulislaw 

> > 2.
> > 
> > We disallow changing main stack size for targets for compatibility sake. What exactly is the problem you are facing?
> 
> In the test used mbed CLI, we face the build error of referencing undefined macro(=  DEFAULT_STACK_SIZE ).
> For example, the value of this macro was 2048 in case of Cortex-A in mbed-os 5.4.
> But this macro was gone by the support of CMSIS5 / RTX5.
>  Therefore, we would appreciate that you modify  DEFAULT_STACK_SIZE  to  2048  in main() function of malloc test.

Does it make sense? If you have any further unclear point on this, please let me know.

toyowata, 2017-09-07:
@TomoYamanaka 

Thank you very much for your update.  Can you share your code in your forked repository, so I can do quick test?
https://github.com/TomoYamanaka/mbed

TomoYamanaka, 2017-09-07:
@toyowata 

Thank you for your comments. Your cooperation will be very helpful.
The code I updated is here.
https://github.com/TomoYamanaka/mbed/tree/Feature_CMSIS5_cortexA

toyowata, 2017-09-07:
@TomoYamanaka 

Can you please sync latest commit from our https://github.com/ARMmbed/mbed-os repo and then apply your commit to the feature_cortex_a branch?

TomoYamanaka, 2017-09-07:
@toyowata 

I synced latest commit from https://github.com/ARMmbed/mbed-os, and
I commited my update to https://github.com/TomoYamanaka/mbed/tree/feature_cortex_a.

Does it make sense? If you have any mistake, please let me know.

TomoYamanaka, 2017-10-17:
Related PR is https://github.com/ARMmbed/mbed-os/pull/5312.

cmonr, 2017-12-14:
Closing since activity has migrated to PR: https://github.com/ARMmbed/mbed-os/pull/5342

closed 2017-12-14 by cmonr