ARMmbed/mbed-os #4816: Eclipse Exporter Enhancements

dlfryar, 2017-07-26:
[image]

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
RTOS awareness and exporter completeness

**Suggested enhancement**
Enhance exporter to complete Eclipse project awareness

**Pros**
Ease of development, feature awareness, references, declarations, IDE hints for code completion, refactoring, debugging

**Cons**
Symbols and target platforms are not resolved after the export

**mbed-cli version:** 
1.2.0

**Expected behavior**
RTOS awareness in Eclipse after exporting source with mbed-cli

**Actual behavior**
1.      ‘mbed export –I eclipse_gcc_arm’ is missing the RTOS defines that allow Eclipse to know about RTOS features.  Additionally, the Eclipse config files are missing exclude paths.
2.      ‘mbed export –I gnuarmeclipse’ is missing the RTOS defines that gives it mbed OS feature awareness.

**Steps to reproduce**
git clone https://github.com/ARMmbed/mbed-os-example-blinky
cd mbed-os-example-blinky
mbed deploy
mbed update
mbed target k64f
mbed toolchain gcc_arm
mbed export -i gnuarmeclipse

-- OR --

mbed export -i eclipse_gcc_arm

dlfryar, 2017-07-26:
@theotherjimmy here's the Eclipse exporter enhancement we discussed.

0xc0170, 2017-07-27:
cc @ilg-ul

ilg-ul, 2017-07-31:
as far as I remember, the exporter simply passes all definitions from the mbed project to the .cproject file, so if the RTOS specific definitions are available, they should be processed by the exporter.

theotherjimmy, 2017-07-31:
That's what I recall as well. I'll have to look at what's going wrong here, if anything.

dlfryar, 2017-08-31:
Retested with latest updates and a clean installation and this seems to no longer be an issue.

closed 2017-08-31 by dlfryar