ARMmbed/mbed-os #4818: Where is the old 'net' library?

sstaub, 2017-07-26:
Where is the old 'net' library gone? It was in the features/unsupported folder.
I need them for multicast feature, which is not supported in mbedos5.
Please bring it back for use with mbed2.

0xc0170, 2017-07-27:
@sstaub It was removed from this repository as it's on developer page. You can use it there (or fetch it locally). For more information (https://developer.mbed.org/handbook/Networking). You can locate there links to the networking libs (mainly EthernetInterface).

 Please use the library from the developer page.

geky, 2017-07-27:
Also a heads up, we have a pr on it's way to get multicast support back up and running: https://github.com/ARMmbed/mbed-os/pull/4820

sstaub, 2017-07-27:
I have seen it, thank you.

0xc0170, 2018-06-18:
Closed as resolved (comments above for more info)

closed 2018-06-18 by 0xc0170