ARMmbed/mbed-os #4823: NUCLEO_L432KC memory definition enhancements should give 16kB (25%) more SRAM

helmut64, 2017-07-28:
The NUCLEO_L432KC target has SRAM1 (46kB) at 0x2000 0000  and SRAM2(16kB) at 0x1000 0000. The SRAM2 is also mapped at 0x2000 C000. At present the SRAM2 is unused within mbed (except for interrupt vectors). 

I recommend to specify in mbed that the SRAM1 is 64kB in size which gives us 16kB (25%) more available SRAM.

Please enhance it. I can do it, however I believe the task belongs to the ST team.

Regards Helmut

BTW: This can also applied to other STM32L4 MCUs. 
- Type:  Enhancement

0xc0170, 2017-07-28:
cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-08-08:
Hi
Idea is good!
Unfortunately, we faced an issue in the ST-Link FW...
Increasing the SRAM1 size will move the initial stack pointer outside from 0x2000 C000 to 0x2001 0000, which should work, but is not currently taken into account by the FW...

jeromecoutant, 2017-08-08:
ST_INTERNAL_REF 36547

helmut64, 2017-08-14:
Hi Jerome,

In the scope of mbed, what has the STLink Firmware to do with SRAM size, the mbed FAT drive just flashes the .bin into the Flash area and done. The initial stack is located on the flash area and not in SRAM.

Yes in mbed we need to adjust the stack, etc.

jeromecoutant, 2017-09-06:
> what has the STLink Firmware to do with SRAM size

As per ARM requirement, Stack Pointer test is used to detect the need to program and run the application coming with drag and drop feature in the virtual mass storage disk.

helmut64, 2017-09-06:
OK than newer ST-Link firmware and changing it in mbed. 25% more memory is worth the work.

helmut64, 2017-10-03:
This enhancement is completed and tested, thanks.
I believe the board page should list the min. STLink version, otherwise this board will not work.
https://os.mbed.com/platforms/ST-Nucleo-L432KC/

jeromecoutant, 2017-10-03:
Done, you can close the issue
Thx

closed 2017-10-03 by helmut64