ARMmbed/mbed-os #4826: Nordic RF52 I2C driver times out too soon for clock-stretching slaves

NeilMacMullen, 2017-07-28:
Mbed 5.5

TARGET_NRF5/i2c_api.c contains this definition...

`// An arbitrary value used as the counter in loops waiting for given event
// (e.g. STOPPED), needed to avoid infinite loops (and not involve any timers
// or tickers).
#define TIMEOUT_VALUE  1000`

which is used to control how long the code waits for various I2C bus operations to complete.

Unfortunately the value of 1000 is too low to cope with slave devices that use clock-stretching so incorrectly generates I2C read and write API errors.

0xc0170, 2017-08-03:
cc @nvlsianpu @anangl

NeilMacMullen, 2017-08-24:
Any thoughts on this?

nvlsianpu, 2017-08-28:
Could you increase this value to 1000000 in test whether it fix the issue.

nvlsianpu, 2017-08-28:
@pan It's possible and suitable to combine a driver with e.g Timer API? Or better it is possible to get global system timer-counter (I mean a value of eg. a us timer running, like time() but in smaller unit)?

NeilMacMullen, 2017-08-29:
@nvlsianpu 
>Could you increase this value to 1000000 in test whether it fix the issue?
I'm currently using a value of 100000 and this works fine for the particular device that was causing problems but I just chose this arbitrarily. :) Do you think 1000000 is more suitable in general?

MarceloSalazar, 2017-09-07:
Bump @pan-

nvlsianpu, 2017-09-15:
@pan- Can you respond to my question (see above)?

pan-, 2017-09-15:
@nvlsianpu It is not desirable to use the C++ Timer API in the C HAL. The HAL should not depends on APIs exposed on higher level. However I believe it is OK to use the timer API from the hal.

0xc0170, 2017-09-21:
> // An arbitrary value used as the counter in loops waiting for given event // (e.g. STOPPED), needed to avoid infinite loops (and not involve any timers // or tickers). #define TIMEOUT_VALUE 1000

Why 1000 was chosen? does it have any relation to the I2c timings ? As above, if slave is clock-stretching, this timeout is small. Would ticker in this case help? How long would we wait?

nvlsianpu, 2017-11-16:
@NeilMacMullen: Dose this issue is solved by nRFx: Use us ticker for I2C timeout #5187?
I will close this issue in the next week if I don't receive any negative response.

NeilMacMullen, 2017-11-17:
Yes- seems  to be working well for me.  Thanks for addressing this!

closed 2017-11-21 by 0xc0170