ARMmbed/mbed-os #4834: SPI driver reports EvrRtxMutexError -6 (Not allowed in ISR context) on STM32F412ZG

javier-moreno-tridonic-com, 2017-07-31:
### Description
- Type: Bug
- Priority: Major
---------------------------------------------------------------
## Bug

**Target**
STM32F412ZG

**Toolchain:**
GCC_ARM

**Toolchain version:**
```
λ arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]
```

**mbed-cli version:** 
```
λ mbed-cli --version
1.2.0
```

**mbed-os sha:**
8828635da469162cf2854b5287561c663fb96e72 (mbed-os-5.5.2 tag)

**DAPLink version:**

**Expected behavior**
SPI access from main() thread to not be qualified as ISR context

**Actual behavior**
SPI access from main() thread triggers mbed_die(). The last output is:
```
Mutex 0x200011d4 error -6: Not allowed in ISR context<\r><\n>
```

**Steps to reproduce**

In our case we only need to try and write to SPI to get the following error:
```
Mutex 0x200011d4 error -6: Not allowed in ISR context<\r><\n>
```

Pausing the execution at that point reveals the following stack trace.
```
us_ticker_read() at us_ticker_32b.c:551 0x8053e4e	
wait_us() at mbed_wait_api_rtos.cpp:43 0x8046e26	
wait_ms() at mbed_wait_api_rtos.cpp:31 0x8046dda	
mbed_die() at mbed_board.c:59 0x8045ca4	
_exit() at mbed_retarget.cpp:765 0x8046994	
__wrap_exit() at mbed_retarget.cpp:816 0x80469b8	
error() at mbed_error.c:42 0x8045f32	
EvrRtxMutexError() at mbed_retarget.cpp:1 048 0x8046c72	
osMutexAcquire() at rtx_mutex.c:461 0x8049fca	
rtos::Mutex::lock() at Mutex.cpp:53 0x8046f00	
mbed::SPI::lock() at SPI.cpp:90 0x801c7ac	
mbed::SPI::write() at SPI.cpp:74 0x801c720	
sst2x_LowLevel_Transfer_Synch() at microchip-sst2x-flash-low-level.cpp:100 0x8058e42	
sst2x_driver::sst2x_CmdReadStatusRegister() at microchip-sst2x-flash-driver.cpp:394 0x8057284	
sst2x_driver::sst2x_ReadData() at microchip-sst2x-flash-driver.cpp:1 943 0x8058208	
lwm2mObjectManagerObject::ReadFromStorageMedia() at lwm2mObjectManager.cpp:1 043 0x8055d42	
lwm2mObjectManagerObject::searchForNewestPageRevision() at lwm2mObjectManager.cpp:358 0x8054f18	
lwm2mObjectManagerObject::lwm2mObjectManagerObject() at lwm2mObjectManager.cpp:293 0x8054cfc	
lwm2mObjectManagerObject::getInstance() at lwm2mObjectManager.cpp:63 0x8054726	
main() at main.cpp:278 0x80088c8	
```

Same code was running ok on mbedOS 5.3.4 

If we comment out (just to try out) the mutex usage from the SPI  driver like so:

```
void SPI::lock() {
   // _mutex->lock();
}

void SPI::unlock() {
    //_mutex->unlock();
}
```

Our code runs OK.

0xc0170, 2017-07-31:
@javier-moreno-tridonic-com This is not connected to IRQ mode, see the line https://github.com/ARMmbed/mbed-os/blob/036778b64ee5f12d70534c0e9f61418987771a38/rtos/rtx5/TARGET_CORTEX_M/rtx_mutex.c#L460

Mutex cannot be acquired or released if interrupts are disabled due to svc calls (more details are provided here when we discuss this https://github.com/ARMmbed/mbed-os/pull/2391)

0xc0170, 2017-07-31:
Another similar thread https://github.com/ARM-software/CMSIS_5/issues/171 regarding this topic

cc @RobertRostohar @JonatanAntoni

javier-moreno-tridonic-com, 2017-08-01:
Thanks for the heads up @0xc0170. Digging a little bit further I saw that the problematic piece of code is an own developed external flash driver using **core_util_critical_section_enter()** and **core_util_critical_section_exit()** as you suggested.

Unfortunately the colleague that implemented that part is on holiday, so I lack the knowledge for the moment, to discuss further what could be a more elegant solution than bluntly commenting out the mutex usage.

I'll close the issue but, would it be ok to reopen it once I have more insight of our driver design? Then we could get some pointers about how to implement it more "rtx5 friendly"

Or is other channel for such discussions preferred?

0xc0170, 2017-08-02:
> I'll close the issue but, would it be ok to reopen it once I have more insight of our driver design? Then we could get some pointers about how to implement it more "rtx5 friendly"
Or is other channel for such discussions preferred?

GitHub should be good for this.

> external flash driver using core_util_critical_section_enter() and core_util_critical_section_exit() as you suggested.

Needs to be looked at why it requires. In this case, it might feasible to overwrite locking methods in SPI for this flash object, if the SPI in this case is protected already in the critical section (be sure that this flash object is accessed safely)

javier-moreno-tridonic-com, 2017-08-03:
>  it might feasible to overwrite locking methods in SPI for this flash object

That's what I actually did for the moment. I inherited SPI in a new class that redefines lock and unlock as empty methods and used the "new"  class in the driver. So far so good.

 I'l close the issue now and reopen once my colleague is back. Thank you for your help :)

closed 2017-08-03 by javier-moreno-tridonic-com