ARMmbed/mbed-os #4845: [nRF51822, BLE]: Set whitelist for BLE module on nRF51822

iotvietmember, 2017-08-02:
### Description
- Type: Question
- Related issue: `#whitelist`

---------------------------------------------------------------
## Question

**How to set white list for advertising procedures with BLE module on nRF51822?**

I am trying to set white list for Advertising BLE on nRF51822. But when I was unsuccessful. 
ble.gap().startAdvertising return error 3  = BLE_ERROR_PARAM_OUT_OF_RANGE

Code:
```
......
/* Setup advertising. */
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::BREDR_NOT_SUPPORTED | GapAdvertisingData::LE_GENERAL_DISCOVERABLE);
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::COMPLETE_LIST_16BIT_SERVICE_IDS, (const uint8_t *)uuid16_list, sizeof(uuid16_list));
    ble.gap().accumulateAdvertisingPayload(GapAdvertisingData::COMPLETE_LOCAL_NAME, (uint8_t *)DEVICE_NAME, sizeof(DEVICE_NAME));
    ble.gap().accumulateScanResponse(GapAdvertisingData::COMPLETE_LOCAL_NAME,(const uint8_t *)DEVICE_NAME, sizeof(DEVICE_NAME));
    ble.gap().setAdvertisingType(GapAdvertisingParams::ADV_CONNECTABLE_UNDIRECTED);
    ble.gap().setAdvertisingInterval(200); 

    /* Using WhiteList for Adv*/   
    PRINTF("(BLE) Max WhiteList size: %d \r\n", ble.gap().getMaxWhitelistSize());    //=8

    const BLEProtocol::AddressBytes_t mobile_ble_address = {0x00, 0x11, 0x22, 0x33, 0x44, 0x44};
    
    BLEProtocol::Address_t address;
    address.type = BLEProtocol::AddressType::RANDOM_STATIC;
    for(int i = 0; i < 6; ++i) {
      address.address[i] = mobile_ble_address[i];
    }

    Gap::Whitelist_t whitelist;    
    whitelist.size = 1;
    whitelist.capacity = 1;
    whitelist.addresses = &address;
    
    error = ble.gap().setWhitelist(whitelist);    //not error
    if (error!=0){
        PRINTF("BLE.setWhitelist with error: %d \r\n", error);   
    }
    
    PRINTF("Adv policy mode (before): %d\r\n", ble.gap().getAdvertisingPolicyMode());  //=0
    error = ble.gap().setAdvertisingPolicyMode(Gap::ADV_POLICY_FILTER_ALL_REQS);      //not error
    if (error!=0){
        PRINTF("BLE.setAdvertisingPolicyMode with error: %d \r\n", error);
    }
    
    PRINTF("Adv policy mode (after): %d\r\n", ble.gap().getAdvertisingPolicyMode()); //=3
    
    /* Start advertising */
    error = ble.gap().startAdvertising();    //has error 3 = BLE_ERROR_PARAM_OUT_OF_RANGE
    if (error!=0){
        PRINTF("BLE.startAdvertising with error: %d \r\n", error);
    }
```

pan-, 2017-08-02:
@iotvietmember In your advertising data, you set the flag `LE_GENERAL_DISCOVERABLE`. If you use the whitelisting feature, your device is **not** in the general discoverable mode. Remove this flag and it will work.

iotvietmember, 2017-08-02:
@pan- Your suggestion is useful. Thank you

closed 2017-08-02 by iotvietmember