ARMmbed/mbed-os #4856: CLI compiling options

mapellil, 2017-08-04:
### Description
- Type: Question
- Related issue: 
- Priority: Minor

---------------------------------------------------------------

**Target**
NUCLEO-F401RE

**Toolchain:**
GCC_ARM Mbed CLI
-----------------------------------------------------------------

## Question

**From the CLI how to change the the default option -mfloat-abi=softfp for the NUCLEO-F401RE (M4) defined in the gcc.py? More in general how to override from CLI the default compiling options defined in the gcc.py ?   **

0xc0170, 2017-08-04:
cc @theotherjimmy

theotherjimmy, 2017-08-04:
@mapellil Why do you need to change `-mfloat-abi=softfp`? What do you need to change it to? 

> More in general how to override from CLI the default compiling options defined in the gcc.py ? 

For general flags, not related to cores, you can create your own build profile. Documentation here: https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/toolchain_profiles/

mapellil, 2017-08-07:
Why not? as the processor i'm using support both sw and hw floating point calculation. More specifically i need to link an external library compiled with hw fp while the CLI impose sw fp so the linker complains. I need to change -mfloat-abi=softfp to -mfloat-abi=hard
thanks, regards

0xc0170, 2017-08-07:
> For general flags, not related to cores, you can create your own build profile. Documentation here: https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/toolchain_profiles/

Does this answer your question?

mapellil, 2017-08-07:
Unfortunately no, as i need to change compiler option you define as "related to core". Adding the option -mfloat-abi=hard to an existing profiles as a "debug" or to a custom defined profile lead to the same result:
a wrong compiler string containing both the options (the -mfloat-abi=softfp defined in gcc.py and the -mfloat-abi=hard defined by me). While evaluating the string the gcc consider only the last defined option that is unfortunately the softfp.
regards

theotherjimmy, 2017-08-07:
> while the CLI impose sw fp

Just to make sure that we are all on the same page here: `-mfloat-abi=softfp` uses the hardware floating point unit, with the software floating point calling convention. Please see https://gcc.gnu.org/onlinedocs/gcc/ARM-Options.html for more information. From my experience there is no difference in performance between `softfp` and `hard` and `softfp` code can link with `soft` code.

My recommendation is to re-compile the library with `-mfloat-abi=softfp`.

If you can't do that, then I can help you modify the tools to meet your needs, but understand that this will only impact your local development and will _not_ work with the online compiler.  For that [unsupported] workaround apply this patch:
```diff
diff --git a/tools/toolchains/gcc.py b/tools/toolchains/gcc.py
index 2eb65dc38..fbe80804e 100644
--- a/tools/toolchains/gcc.py
+++ b/tools/toolchains/gcc.py
@@ -66,7 +66,7 @@ class GCC(mbedToolchain):
         # FPU handling, M7 possibly to have double FPU
         if target.core == "Cortex-M4F":
             self.cpu.append("-mfpu=fpv4-sp-d16")
-            self.cpu.append("-mfloat-abi=softfp")
+            self.cpu.append("-mfloat-abi=hard")
         elif target.core == "Cortex-M7F":
             self.cpu.append("-mfpu=fpv5-sp-d16")
             self.cpu.append("-mfloat-abi=softfp")
```

mapellil, 2017-08-07:
i've no performance issues but library compatibility issue. Unfortunately I can't recompile the library as i do not have the src code. Nevertheless force to compile all the libs with -mfloat-abi=softfp option seems to me too limiting regardless of the reasons why the users want to do this. What's the compelling reason to hard code this gcc option? Is there any documentation listing this gcc options limitations and eventually others?
-            self.cpu.append("-mfloat-abi=softfp")
+            self.cpu.append("-mfloat-abi=hard")
This is what i did but seemed to me a just a workaround preventing the pubblication, not a clean solution; this is the reason of this thread.
thanks

mapellil, 2017-08-09:
i've a proposal: why not to consider to override the -mfloat-abi=softfp default defined in the gcc.py if the user specify the same option as -mfloat-abi=hard in the user_opt.json ?
thanks

bmcdonnell-ionx, 2017-12-12:
This issue should be handled for the exporters, as well as the build tools.

e.g. Currently I'm using the Eclipse-based NXP MCUXpresso IDE. It exports with the soft ABI option. I can change it to hard in the project properties in MCUXpresso, but then I'd have to redo that tweak every time I export (i.e. every time I add or update a library with mbed-cli - such as updating the mbed-os library).

theotherjimmy, 2018-07-03:
We have not plans to support `-mfloat-abi=hard`, as it can easily fragment the ecosystem's pre-compiled binaries (.o, .a).

closed 2018-07-03 by theotherjimmy