ARMmbed/mbed-os #4862: Starting with 5.5.4, gnuarmeclipse linker fails with mbed-client

amq, 2017-08-06:
## Bug

**Target**
All

**Toolchain:**
GCC_ARM

**Toolchain version:**
6 2017-q1-update

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
4c256f04596179699c4f14b6863b07cc024ca9be

**Expected behavior**
Can build the client example

**Actual behavior**
Client example fails with:

```
Invoking: Cross ARM C++ Linker
arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -O0 -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin -funsigned-char -fno-delete-null-pointer-checks -fomit-frame-pointer -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers  -g3 -T "Debug\linker-script-debug.ld" -Xlinker --gc-sections -Xlinker --wrap=main -Xlinker --wrap=_malloc_r -Xlinker --wrap=_free_r -Xlinker --wrap=_realloc_r -Xlinker --wrap=_memalign_r -Xlinker --wrap=_calloc_r -Xlinker --wrap=exit -Xlinker --wrap=atexit -Wl,-Map,"mbed-client-example.map" -DMBED_DEBUG -DMBED_TRAP_ERRORS_ENABLED=1 -Wl,-n -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -Wl,--end-group -o "mbed-client-example.elf"...
./mbed-client/source/m2mtlvserializer.o: In function `M2MTLVSerializer::serialize_TLV_binary_int(M2MResourceBase const*, unsigned char, unsigned short, unsigned char*&, unsigned long&)':
Debug/../mbed-client/source/m2mtlvserializer.cpp:207: undefined reference to `common_write_64_bit'
./mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.o: In function `M2MConnectionHandlerPimpl::send_receive_event()':
Debug/../mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp:92: undefined reference to `eventOS_event_send'
./mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.o: In function `M2MConnectionHandlerPimpl::M2MConnectionHandlerPimpl(M2MConnectionHandler*, M2MConnectionObserver&, M2MConnectionSecurity*, M2MInterface::BindingMode, M2MInterface::NetworkStack)':
Debug/../mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp:141: undefined reference to `eventOS_scheduler_mutex_wait'
Debug/../mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp:143: undefined reference to `eventOS_event_handler_create'
Debug/../mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp:145: undefined reference to `eventOS_scheduler_mutex_release'
./mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.o: In function `M2MConnectionHandlerPimpl::send_dns_event()':
Debug/../mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp:193: undefined reference to `eventOS_event_send'
./mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.o: In function `M2MConnectionHandlerPimpl::send_data(unsigned char*, unsigned short, sn_nsdl_addr_*)':
Debug/../mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp:410: undefined reference to `eventOS_event_send'
./mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.o: In function `M2MConnectionHandlerPimpl::claim_mutex()':
Debug/../mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp:666: undefined reference to `eventOS_scheduler_mutex_wait'
./mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.o: In function `M2MConnectionHandlerPimpl::release_mutex()':
Debug/../mbed-client/mbed-client-classic/source/m2mconnectionhandlerpimpl.cpp:671: undefined reference to `eventOS_scheduler_mutex_release'
./mbed-client/mbed-client-classic/source/m2mtimerpimpl.o: In function `tasklet_func':
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:49: undefined reference to `eventOS_scheduler_mutex_wait'
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:54: undefined reference to `eventOS_scheduler_mutex_release'
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:65: undefined reference to `eventOS_scheduler_mutex_release'
./mbed-client/mbed-client-classic/source/m2mtimerpimpl.o: In function `M2MTimerPimpl::M2MTimerPimpl(M2MTimerObserver&)':
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:81: undefined reference to `ns_hal_init'
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:82: undefined reference to `eventOS_scheduler_mutex_wait'
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:84: undefined reference to `eventOS_event_handler_create'
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:91: undefined reference to `eventOS_scheduler_mutex_release'
./mbed-client/mbed-client-classic/source/m2mtimerpimpl.o: In function `M2MTimerPimpl::~M2MTimerPimpl()':
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:103: undefined reference to `eventOS_scheduler_mutex_wait'
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:112: undefined reference to `eventOS_scheduler_mutex_release'
./mbed-client/mbed-client-classic/source/m2mtimerpimpl.o: In function `M2MTimerPimpl::start()':
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:147: undefined reference to `eventOS_event_timer_request'
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:152: undefined reference to `eventOS_event_timer_request'
./mbed-client/mbed-client-classic/source/m2mtimerpimpl.o: In function `M2MTimerPimpl::cancel()':
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:161: undefined reference to `eventOS_event_timer_cancel'
./mbed-client/mbed-client-classic/source/m2mtimerpimpl.o: In function `M2MTimerPimpl::start_still_left_timer()':
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:214: undefined reference to `eventOS_event_timer_request'
Debug/../mbed-client/mbed-client-classic/source/m2mtimerpimpl.cpp:219: undefined reference to `eventOS_event_timer_request'
./mbed-client/mbed-client-c/source/sn_grs.o: In function `sn_grs_destroy':
Debug/../mbed-client/mbed-client-c/source/sn_grs.c:62: undefined reference to `ns_list_remove_'
Debug/../mbed-client/mbed-client-c/source/sn_grs.c:61: undefined reference to `ns_list_get_next_'
./mbed-client/mbed-client-c/source/sn_grs.o: In function `sn_grs_init':
Debug/../mbed-client/mbed-client-c/source/sn_grs.c:139: undefined reference to `sn_coap_protocol_init'
./mbed-client/mbed-client-c/source/sn_grs.o: In function `sn_grs_get_next_resource':
Debug/../mbed-client/mbed-client-c/source/sn_grs.c:242: undefined reference to `ns_list_get_next_'
./mbed-client/mbed-client-c/source/sn_grs.o: In function `sn_grs_put_resource':
[... 63 lines cut ...]
./mbed-client/mbed-client-c/source/sn_nsdl.o: In function `sn_nsdl_internal_coap_send':
Debug/../mbed-client/mbed-client-c/source/sn_nsdl.c:785: undefined reference to `sn_coap_builder_calc_needed_packet_data_size_2'
Debug/../mbed-client/mbed-client-c/source/sn_nsdl.c:798: undefined reference to `sn_coap_protocol_build'
./mbed-client/mbed-client-c/source/sn_nsdl.o: In function `sn_nsdl_build_response':
Debug/../mbed-client/mbed-client-c/source/sn_nsdl.c:1654: undefined reference to `sn_coap_build_response'
./mbed-client/mbed-client-c/source/sn_nsdl.o: In function `sn_nsdl_alloc_options_list':
Debug/../mbed-client/mbed-client-c/source/sn_nsdl.c:1662: undefined reference to `sn_coap_parser_alloc_options'
./mbed-client/mbed-client-c/source/sn_nsdl.o: In function `sn_nsdl_release_allocated_coap_msg_mem':
Debug/../mbed-client/mbed-client-c/source/sn_nsdl.c:1671: undefined reference to `sn_coap_parser_release_allocated_coap_msg_mem'
./mbed-client/mbed-client-c/source/sn_nsdl.o: In function `sn_nsdl_set_retransmission_parameters':
Debug/../mbed-client/mbed-client-c/source/sn_nsdl.c:1680: undefined reference to `sn_coap_protocol_set_retransmission_parameters'
collect2.exe: error: ld returned 1 exit status
```

**Steps to reproduce**

1. Export the project for gnuarmeclipse
2. Import it in Eclipse
3. Try building, should fail
4. Change the mbed-os library to 5.5.3, re-export, re-import
5. Try building, should work

I tried starting the project completely from scratch, in a different location, with all libraries removed and installed using `mbed update`.

Maybe it would be more appropriate to create this issue in https://github.com/ARMmbed/mbed-client or https://github.com/ARMmbed/mbed-os-example-client, but I believe the problem is with the exporter, because everything worked with 5.5.3 and nothing changed in those two repos.

amq, 2017-08-07:
Tried some more examples now:
https://github.com/ARMmbed/mbed-os-example-blinky - ok
https://github.com/ARMmbed/mbed-os-example-cellular - fails
https://github.com/ARMmbed/mbed-os-example-fat-filesystem - ok

theotherjimmy, 2017-08-07:
I'll take a look, but I'm a bit swamped right now. I'm adding the tracking label so that I will have this issue is my queue for later.

amq, 2017-08-11:
Just to confirm, the issue is also present in 5.5.5

amq, 2017-08-26:
Reverting https://github.com/ARMmbed/mbed-os/pull/4783 solves the issue for me.

amq, 2017-11-23:
@theotherjimmy any update?

theotherjimmy, 2017-11-27:
@amq No, I have not had a chance to take a look at this yet. It looks like it might be related to `"target.features_add" : ["LWIP", "COMMON_PAL"]` in `mbed_app.json`. I'll see if I can reduce it to a minimal breaking example.

amq, 2018-01-28:
Just tested 5.7.4 with the projects which were failing:
https://github.com/ARMmbed/mbed-os-example-client
https://github.com/ARMmbed/mbed-os-example-cellular

... and the issue seems to be gone. The linker command (`arm-none-eabi-g++ -mcpu...`) includes `FEATURE_LWIP` and `FEATURE_COMMON_PAL` again.

Other versions:
mbed-cli: 1.3.0
gcc-arm-none-eabi: 6-2017-q2-update
GNU make: 4.1

closed 2018-01-28 by amq