ARMmbed/mbed-os #4865: BLE UARTService - nRF52-DK unable to send more than 140 bytes of data

niranjanraoh, 2017-08-07:
### Description

I am trying to send a string of length 450 bytes, using BLE UARTService on nRF52-DevKit

I only receive 140 bytes (20 x 7) of data every time I try to send a string whose length is 450 bytes.
I used nRF Connect android app to receive data, I also had the same result when I used my python BLE client (using BluePy)
This 140 bytes is always constant. Every time I call the send function it only sends 14bytes of data
I tried splitting the string into 20-byte chunks and transmitting them. I still receive only 140 bytes.

**Target**
NRF52_DK

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc-4.9.3  

-----------------------------------------------------------------

pan-, 2017-08-08:
@niranjanraoh The UART service is unreliable, data are not guaranteed to arrive to the peer (use of notification) and data are not guaranteed to be buffered by the stack ([see](https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BLE/ble/services/UARTService.h#L93-L94)). 

I would advise you to split your string into smaller chunks and send one chunk at once; with a delay between each transmission.

niranjanraoh, 2017-08-08:
@pan- thank you, it works when I transmit the data in smaller chunks

closed 2017-08-09 by pan-