ARMmbed/mbed-os #4866: Query: Exclusive Access

deepikabhavnani, 2017-08-07:
**__EXCLUSIVE_ACCESS** Macro is set in "core_cm.h" file considering the ARM architecture. 

```
#ifndef __EXCLUSIVE_ACCESS
#if   ((__ARM_ARCH_7M__      == 1U) || \
       (__ARM_ARCH_7EM__     == 1U) || \
       (__ARM_ARCH_8M_MAIN__ == 1U))
#define __EXCLUSIVE_ACCESS      1U
#else
#define __EXCLUSIVE_ACCESS      0U
#endif
#endif
```
**EXCLUSIVE_ACCESS** macro is set in "mbed_critical.c" file.

`#define EXCLUSIVE_ACCESS (!defined (__CORTEX_M0) && !defined (__CORTEX_M0PLUS))`

Shouldn't we use single macro, as it looks both of them are for same purpose. Or any specific reason to have this redefined in *.c file?

@sg- @0xc0170 @bulislaw

0xc0170, 2017-08-08:
> Shouldn't we use single macro, as it looks both of them are for same purpose. Or any specific reason to have this redefined in *.c file?

I dont recall if we did not have it previously in the older code base. But would make sense to switch to cmsis one 👍 

Looking at this further, they added as well atomic_ functions

geky, 2017-08-08:
If it's prefixed with "__", does that mean it's intended to be private?

0xc0170, 2017-08-08:
> If it's prefixed with "__", does that mean it's intended to be private?

I would expect so, but if you look at the macros defined in cmsis headers, that we even use, all of them start and end with `__` .

pan-, 2017-08-09:
Just a remainder any identifier that begin by an underscore followed by  another underscore or an uppercase are **reserved** and should not be used (C99: 7.1.3). 

Identifier beginning with an underscore are also reserved for use as identifier of file scope elements (C99: 7.1.3). Yes, `_private_variable_name` is enforced by the language.

closed 2017-08-23 by deepikabhavnani