ARMmbed/mbed-os #4867: Incremental compiles for config system

geky, 2017-08-07:
This is a pretty big feature request, but it would be a big improvement for users using the config system. I wanted to get it tracked somewhere in case of a rainy day.

As is, any change to a configuration file (mbed_app.json or mbed_lib.json) requires a full recompile. This takes 1min 28sec on my system with only the default profile. This is exacerbated by the fact that most config changes are small tweaks that occur repeatedly to reach a goal, and these tweaks are often carried out by a new user trying to get their system working.

Incrementally compiling only the files actually depend on the modified configuration options would greatly improve the user experience with the config system.

Unfortunately this isn't very straightforward with how the tools currently handle mbed_config.h. Bouncing ideas around it seems the best case would be to create the dependency information ourselves, either by scanning non-recompiled files for config options at compile time, or by storing config dependency information in the BUILD directory.

pan-, 2017-08-10:
> Incrementally compiling only the files actually depend on the modified configuration options would greatly improve the user experience with the config system.

How would you determine the dependency between a configuration option and a file ? IIRC the computed configuration is applied globally to all the files of the project. 

From my understanding, your issue can be solved if the build system embed: a concept of modules, dependency between modules and definition of what is exported by a module (header files, public configuration definition, ...). That would also help user to export reusable precompiled library.

theotherjimmy, 2017-08-10:
@geky Thanks for creating this issue to track the feature request. 

@pan- I think what you are suggesting implies a new restriction on the config system:  that  files within the directory tree rooted at an `mbed_lib.json` file can only use configuration variables from that file and global target configuration (but that's debatable too). That restriction would limit the scope of configuration, allowing the module like behavior, and limit the scope of configuration related recompiles.

> How would you determine the dependency between a configuration option and a file ?

Cop out answer: We scan the files we are compiling, and the header files, for the configuration variable string and mark any file that contains it as "dirty", forcing any file that depends on it to be recompiled. This does not account for using macro concatenation to "dynamically" create new macros and expand them.

pan-, 2017-08-11:
> I think what you are suggesting implies a new restriction on the config system: that files within the directory tree rooted at an mbed_lib.json file can only use configuration variables from that [...]

Not exactly, at the moment an mbed OS application is built as a monolithic block of code where every entity (header file, configuration, ...) is public. For instance, nothing prevents the user to start using private headers and function even if it should be private. We've seen that enough with the RTX update. Another visible limitation of the system is the recompilation of all source files whenever a configuration parameter change. 

It would be possible to overcome those limitations if we had the possibility to instruct the build system that a group of source files form a module (might be files within the directory tree rooted at an `mbed_lib.json` file). The processing of a module would result in the export of object files compiled, a list of public header files and the exported configuration parameters. These entities would be reused by modules depending on the one processed. 

From the list of modules dependencies it would be possible to construct an oriented graph of module dependencies and deduce which module needs to be recompiled if a configuration parameter change.

> We scan the files we are compiling, and the header files, for the configuration variable string and mark any file that contains it as "dirty", forcing any file that depends on it to be recompiled.

`#include` behavior should also be mimicked before the scan for the configuration. It means that compiler include path should be known. I'd be curious to see how much time it takes to search for a pattern on millions of line of code.

> This does not account for using macro concatenation to "dynamically" create new macros and expand them.

That's unfortunate, configuration parameter are good candidate for such macro given they are prefixed.

theotherjimmy, 2017-08-11:
> #include behavior should also be mimicked before the scan for the configuration. It means that compiler include path should be known. I'd be curious to see how much time it takes to search for a pattern on millions of line of code.

That's easy: if we have a `.d` file, we know it's dependencies. If we don't have a `.d` file, then we have not compiled yet. As for speed....
```
# time rg OS_THREAD_LIBSPACE_NUM
rtos/rtx5/mbed_rtx_handlers.c
52:        // Standard C/C++ library libspace not available: increase OS_THREAD_LIBSPACE_NUM

rtos/rtx5/TARGET_CORTEX_M/RTX_Config.c
53:      // Standard C/C++ library libspace not available: increase OS_THREAD_LIBSPACE_NUM

rtos/rtx5/TARGET_CORTEX_M/rtx_lib.c
544:static uint32_t os_libspace[OS_THREAD_LIBSPACE_NUM+1][LIBSPACE_SIZE/sizeof(uint32_t)] \
548:static osThreadId_t os_libspace_id[OS_THREAD_LIBSPACE_NUM] \
573:    return (void *)&os_libspace[OS_THREAD_LIBSPACE_NUM][0];
577:  for (n = 0U; n < OS_THREAD_LIBSPACE_NUM; n++) {
587:  if (n == OS_THREAD_LIBSPACE_NUM) {

rtos/rtx5/TARGET_CORTEX_M/RTX_Config.h
376:#define OS_THREAD_LIBSPACE_NUM      4
378:#define OS_THREAD_LIBSPACE_NUM      OS_THREAD_NUM
real    0.069228123s
```
It should not be too bad. 120ms (a more reasonable number, we're using python here) is less than most compile jobs, so it should be a boon.

deepikabhavnani, 2018-12-28:
@geky - Seems good time to re-look this

loverdeg-ep, 2019-09-16:
@ARMmbed/mbed-os-tools  Any status on this or is it stale enough 
to close?

0xc0170, 2019-09-17:
@loverdeg-ep I would assume the feature like this can be closed as won't fix (not that it won't be fixed somehow - using different tool or so, but changing the current config system to add this is not feasable). 

@ARMmbed/mbed-os-tools team to provide more details and suggest what shall we do.

mark-edgeworth, 2019-09-17:
We're aware of this request but don't have time at the moment to look at it. We're looking to make larger scale improvements over time. No timescale on this yet though.

closed 2019-09-17 by mark-edgeworth