ARMmbed/mbed-os #4870: Facing error: KL46 :RTX error code: 0x00000004, task ID: 0x1FFFF160

samp12345, 2017-08-08:
Hi,
I am using Freescale KL46 for interfacing with few SPI devices.
During this, I get the issue "RTX error code: 0x00000004, task ID: 0x1FFFF160"
I am using online mbed RTOS compiler directly 
The error is consistent.
Please suggest,

0xc0170, 2017-08-08:
The code 0x4 for the error indicates osRtxErrorClibSpace.. What version of mbed OS are you using? There has been recently a fix  to provide heap for allocating mutex dynamically (first pool then heap). It is part of 5.5.4 : https://github.com/ARMmbed/mbed-os/pull/4731 

cc @bulislaw

samp12345, 2017-08-08:
As i use the online compiler, it should be latest i trust. Still i have issues, it is repeatable.
Yes, what you said is correct, with same code i was not having any issues six months back.

0xc0170, 2017-08-08:
> As i use the online compiler, it should be latest i trust. Still i have issues, it is repeatable.

Just to be certain, please check revision in the online compiler (or mbed.h version macro, if it states 5.5.4). You should see these  changes in there https://github.com/ARMmbed/mbed-os/pull/4731/files

If still fails even with the latest mbed OS, then we would request a simple codesnippet to reproduce the the error. This would help us understand the issue and fix it quicker.

samp12345, 2017-08-09:
Thank you, I updated with latest revisions, still i have issues. 
I didn't implement muted in this code.
Basically i have RFM22B which is talking toit over SPI. SPI is not shared with any other device.
When i fill the fifo of Rfm22b, this error occurs.
Rfm22b is separately tested, it is good.

bulislaw, 2017-08-14:
Can you get a backtrace or try to debug it a bit more?

samp12345, 2017-08-16:
Apologies for the delayed response.  I am trying to debug more to have additional information,
Soon 'll get back to you with more details.

0xc0170, 2018-06-21:
Hi @samp12345 , please reopen with update

closed 2018-06-21 by 0xc0170