ARMmbed/mbed-os #4871: Compilation error of the BLUEPILL_F103C8 target

vznncv, 2017-08-08:
Mbed doesn't compile any project for STM32F103C8 microcontroller (**BLUEPILL_F103C8** board).

The error:
```
Scan: .
Scan: mbed
Scan: env
Compile [  0.4%]: FlashIAP.cpp
Compile [  0.7%]: CAN.cpp
Compile [  1.1%]: Ethernet.cpp
Compile [  1.4%]: BusIn.cpp
[Error] mbed_rtx.h@101,2: #error "INITIAL_SP is not defined for this target in the mbed_rtx.h file"
[ERROR] In file included from ./mbed-os/rtos/rtx5/mbed_rtx_conf.h:25:0,
                 from ./mbed-os/rtos/rtx5/TARGET_CORTEX_M/RTX_Config.h:33,
                 from ./mbed-os/rtos/rtx5/TARGET_CORTEX_M/rtx_evr.h:33,
                 from ./mbed-os/rtos/rtx5/TARGET_CORTEX_M/rtx_lib.h:37,
                 from ./mbed-os/rtos/mbed_rtos_storage.h:43,
                 from ./mbed-os/rtos/Mutex.h:28,
                 from ./mbed-os/platform/PlatformMutex.h:24,
                 from ./mbed-os/drivers/BusIn.h:21,
                 from ./mbed-os/drivers/BusIn.cpp:16:
./mbed-os/targets/TARGET_STM/mbed_rtx.h:101:2: error: #error "INITIAL_SP is not defined for this target in the mbed_rtx.h file"
 #error "INITIAL_SP is not defined for this target in the mbed_rtx.h file"
  ^

[mbed] ERROR: "/usr/bin/python" returned error code 1.
```

Reason: in the file `targets/TARGET_STM/mbed_rtx.h` the `INITIAL_SP` macros definition is missed for `TARGET_STM32F103C8`.

### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
BLUEPILL_F103C8

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
4a99b3b

**Expected behavior**
Compilation without error.

**Actual behavior**
Compilation fails.

**Steps to reproduce**
```
mbed new test_stm32f103c8
cd test_stm32f103c8
mbed compile --target BLUEPILL_F103C8 --toolchain GCC_ARM
```

0xc0170, 2017-08-08:
@vznncv Hello, this target does not support mbed OS 5, thus the failure

cc @bcostm @adustm @LMESTM @jeromecoutant

vznncv, 2017-08-08:
@0xc0170, Hello

> @vznncv Hello, this target does not support mbed OS 5, thus the failure

But you have files for this target (https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32F1/TARGET_BLUEPILL_F103C8), and I compiled some programs for it, when I manually added `INITIAL_SP` macro into `mbed_app.json`:
```
{
    "target_overrides": {
        "BLUEPILL_F103C8": {
            "target.macros_add": [
                "INITIAL_SP=0x20005000UL"
            ]
        }
    }
}
``

adustm, 2017-08-08:
Hello @0xc0170 

This target is not supported by ST team. 
We don't know who is maintaining this target.

Kind regards

jeromecoutant, 2017-08-08:
Note that STM32F103C8 will be too light in term of FLASH and RAM for OS5 support

0xc0170, 2017-08-08:
> Note that STM32F103C8 will be too light in term of FLASH and RAM for OS5 support

As stated above.  This target is not officially supported. IT was ported to mbed 2 only.

@vznncv You can add the support and it test it locally. It was already asked here https://github.com/ARMmbed/mbed-cli/issues/377, https://github.com/ARMmbed/mbed-os/issues/2484 and https://github.com/ARMmbed/mbed-os/pull/4255/files.

cc @wassfila @zgoda

closed 2017-08-11 by vznncv

wassfila, 2017-08-13:
@jeromecoutant, @adustm  other reasons could be fine to discard this target but regarding Flash, please consider using "--specs=nano.specs" I have the mbed os 5 happily running with around 35 KB with my own branch [here](https://github.com/wassfila/STM32_IoT_HelloWorld/tree/master/08_mbed_cli_os_BluePill_Target) I just had testing issues, I'm not used to the environment that prevented my pull request to be accepted, if someone is willing to spend the testing efforts I can help him.
For the RAM, it also depend, it's fine as long as you don't include USB or other heavy stuff.
Keep in mind this is the world's cheapest target by an order of magnitude which make it relevant for IoT projects which cost is a decisive factor.
I already have more than 10 Hello world samples with different sensors running fine on this ultra low cost target, I do not know why people are ignoring the potential of this target.

wassfila, 2017-08-13:
@vznncv here's the [commit](https://github.com/wassfila/mbed-os/commit/3d3dfb1800ad5416e9b3421d0515044f1a288578) with the change you're looking for. As mentionned above, I abandoned due to lack of testing efforts, but if you're interested feel free to use that fork in any way, let me know if you need more info.