ARMmbed/mbed-os #4872: Updating gnu version to gnu++11

theamirocohen, 2017-08-08:
### Description
- Type: Enhancement
- Priority: Minor
----------------------------------------------------------------
## Enhancement
Due to project requirements I wanted to update mbed-os gnu version from gnu++98 to gnu++11, in order to use C++11 functionalities.

amq, 2017-08-14:
It has been [recommended](https://github.com/ARMmbed/mbed-os/pull/4864) to use custom build profiles, but it doesn't apply to exporting, so it looks like there is no elegant way to enable C++11 for Eclipse currently.

theotherjimmy, 2017-08-14:
@amq Profiles do apply to exportings, just not the GNU ARM Eclipse exporter. You could use the `eclipse_gcc_arm` exporter with a profile if you wanted. The `--profile` argument is accepted.

amq, 2017-08-15:
I've noticed that [PAL](https://github.com/ARMmbed/pal) is not compatible with C++11, hopefully the new client doesn't need it or something 😄

pan-, 2017-08-25:
@amq Just for the record: Looking at the fix, this has nothing to do with C++11. The behavior of left shifting a negative value has never been defined by any C or C++ standards. From the beginning, even in C++ 98, the result of the expression `(-1 << SOME_CONSTANT)` was undefined (from the standard perspective).

bmcdonnell-ionx, 2018-04-16:
I found at least one reason to want to enable C++11 (or higher) not just in user projects, but also within Mbed OS itself: to replace [`mbed::Callback`](https://github.com/ARMmbed/mbed-os/blob/master/platform/Callback.h) with [`std::function`](http://en.cppreference.com/w/cpp/utility/functional/function). `mbed::Callback` is apparently a kludge to work-around the lack of `std::function`. IME, it is not friendly to work with.

`mbed::Callback` is admittedly a stand-in for `std::function`, per the [Mbed OS Platform documentation ](https://os.mbed.com/docs/v5.8/reference/platform.html): "The Callback class should be familiar to users of the std::function class that C++11 introduced and is available for older versions of C++."

I have C++11 enabled in an Mbed OS-based project. Trying to create `Callback`s from lambda functions is *not fun*. I've only been able to use one or two variable captures, due to the limited number of parameters accepted by `mbed::Callback`. (Presumably the lambda captures get translated to function parameters. I'm not sure where the extras come from.) I created a wrapper class in one case, just so I could `attach` a `std::function` to it.

The template copy-pasta in `mbed::Callback` also impedes understanding from reading the code.

I give props for the work-around, but I think it's had its day. What compilers and IDEs remain nowadays that don't support at least C++11, such that the Mbed team is still maintaining backward compatibility with older versions of C++? Can we please move on?

theotherjimmy, 2018-04-16:
@bmcdonnell-ionx We would need to upgrade to Arm Compiler 6 and IAR EWARM 8. It's not so easy though: IAR EWARM 8 does not support C++03. [This issue](https://github.com/ARMmbed/mbed-os/issues/5507) has more background on what an upgrade to C++11 would entail.

theotherjimmy, 2018-04-16:
Actually, it looks like C++14 would be the standard when we upgrade.

bmcdonnell-ionx, 2018-04-17:
@theotherjimmy, I knew I had seen a more comprehensive issue than this one! Thanks for the link.

Should this issue be closed, as it is superseded by #5507?

idea--list, 2018-12-03:
Hi there,

I am learning C++ right now for an embedded project. I notice the way one should code depends also on the C++ standard. As i see the online mbed compiler still compiles in C++98, which i find quite odd to say the least even as a beginner.

My question is: Is there a way i can change the setup of the online compiler to C+11 or C+14 standards?
If not can i just write my code locally in C++11 without using mbed-os (that seems to be outdated) and just upload a locally compiled bin to my board?

kjbracey-arm, 2018-12-04:
I don't believe you can change the online compiler, but if building locally, there is nothing stopping you from changing the language standard in `tools/toolchains/toolchainname.py`.

We are in the process of updating officially-supported toolchains and language standards, but we can only actively support and test one version of each, giving a total of 3 combinations at the minute - the compiler versions listed in the docs pages with the profiles the profiles you see in `tools/toolchains`.

I think you will have success with reasonable modifications to those profiles, including switching to C++11, C++14 or C11. I believe this will probably work most of the time on most platforms, but it is not properly tested and is unsupported.

theotherjimmy, 2018-12-04:
@kjbracey-arm the c++/c standard is controlled by the build profile.

bmcdonnell-ionx, 2018-12-04:
@idea--list, I've used a modified build profile to [`mbed compile`](https://github.com/ARMmbed/mbed-cli#compiling-your-program) a program with C++11.