ARMmbed/mbed-os #4876: gt: test case summary event not found

gorazdko, 2017-08-08:
# Running mbed test -m XDOT_L151CC -t ARM
   - is unsuccessful in 4a99b3bbbc9d87e0b3c006e50e517d8f47ec1e07
`mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: test specification file 'C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os\BUILD\tests\XDOT_L151CC\ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os\BUILD\tests\XDOT_L151CC\ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 2 devices
mbedgt: processing target 'XDOT_L151CC' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: running 56 tests for platform 'XDOT_L151CC' and toolchain 'ARM'
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: features-frameworks-utest-tests-unit_tests-test_setup_failure
        test case: features-frameworks-utest-tests-unit_tests-test_setup_failure
mbedgt: test on hardware with target id: 035000000388109100000000000000000000000097969902
mbedgt: test suite 'features-frameworks-utest-tests-unit_tests-test_setup_failure' ................... UNDEF in 17.70 sec
        test case: 'features-frameworks-utest-tests-unit_tests-test_setup_failure' ................... ERROR in 17.70 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: tests-mbedmicro-rtos-mbed-queue
        test case: tests-mbedmicro-rtos-mbed-queue
mbedgt: test on hardware with target id: 035000000388109100000000000000000000000097969902
mbedgt: test suite 'tests-mbedmicro-rtos-mbed-queue' ................................................. UNDEF in 18.22 sec
        test case: 'tests-mbedmicro-rtos-mbed-queue' ................................................. ERROR in 18.22 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: mbed-host-test-runner: started`
   - is succsessful in 1af8aef5dc0dd981f23bfe3cfdfb9df8d6e820f9 
`
      mbedgt: test case results: 217 OK
mbedgt: completed in 1339.67 sec
`
### I use:
C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos>mbedgt --version
1.2.5

C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos>mbedhtrun --version
1.1.9

C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos>mbedls --version
1.2.13

C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos>mbed --version
1.2.0

## Expected beahviour
should run successfully in both cases

## Question:
How to run mbed test -m XDOT_L151CC -t ARM successfully in the latest release?

## Additional info
`C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos>mbedls
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| FF1705_L151CC | FF1705_L151CC[0]     | E:          | COM3        | 80800000038811ca00000000000000000000000097969902 | 0243            |
| XDOT_L151CC   | XDOT_L151CC[0]       | F:          | COM13       | 035000000388109100000000000000000000000097969902 | 0243            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+`

0xc0170, 2017-08-09:
cc @mfiore02

0xc0170, 2017-08-09:
It would help if you use verbose move (-v) to see why it is undef for some. I assume based on the output above, that you run all the tests, and only some fail (those 2 shared above? ).

gorazdko, 2017-08-09:
All tests fail for the same reason. Didnt want to spam too much in the comment.

C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os>mbed test -m XDOT_L151CC -t ARM -v
[mbed] Working path "C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os" (program)
[mbed] WARNING: Could not find mbed program in current path "C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os".
[mbed] WARNING: You can fix this by calling "mbed new ." in the root of your program.
---
[mbed] Exec "c:\python27\python.exe -u C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os\tools\test.py -t ARM -m XDOT_L151CC --source C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os --build C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os\BUILD\tests\XDOT_L151CC\ARM --test-spec C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os\BUILD\tests\XDOT_L151CC\ARM\test_spec.json -v" in C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
Building library mbed-build (XDOT_L151CC, ARM)
Scan: mbed-os
[DEBUG] Macros: -DTARGET_XDOT_L151CC -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32L151CC -DTOOLCHAIN_object -D__CMSIS_RTOS -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DDEVICE_RTC=1 -DDEVICE_ANALOGOUT=1 -DTARGET_STM32L1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_PWMOUT=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FAMILY_STM32 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL=1 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DMBED_BUILD_TIMESTAMP=1502268648.07 -DDEVICE_FLASH=1 -DTOOLCHAIN_ARM -DTOOLCHAIN_ARM_STD
Scan: ARM
WWARNING: MBED_ARM_PATH set as environment variable but doesn't existARNING: MBED_ARM_PATH set as environment variable but doesn't exist

WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
WARNING: MBED_ARM_PATH set as environment variable but doesn't exist
Building project rtc_time (XDOT_L151CC, ARM)
Scan: ARM
Scan: rtc_time
[DEBUG] Macros: -DTARGET_XDOT_L151CC -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32L151CC -DTOOLCHAIN_object -D__CMSIS_RTOS -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DDEVICE_RTC=1 -DDEVICE_ANALOGOUT=1 -DTARGET_STM32L1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_SERIAL=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FAMILY_STM32 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DMBED_BUILD_TIMESTAMP=1502268650.97 -DDEVICE_FLASH=1 -DTOOLCHAIN_ARM -DTOOLCHAIN_ARM_STD
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Misc      | 47630 |  2510 | 6284 |
| Subtotals | 47630 |  2510 | 6284 |
+-----------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 8794 bytes
Total RAM memory (data + bss + heap + stack): 8794 bytes
Total Flash memory (text + data + misc): 50140 bytes
Image: BUILD/tests/XDOT_L151CC/ARM/TESTS/mbed_hal/rtc_time/rtc_time.bin

Building project test_setup_failure (XDOT_L151CC, ARM)
Scan: ARM
Scan: test_setup_failure
[DEBUG] Macros: -DTARGET_XDOT_L151CC -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32L151CC -DTOOLCHAIN_object -D__CMSIS_RTOS -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DDEVICE_RTC=1 -DDEVICE_ANALOGOUT=1 -DTARGET_STM32L1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_SERIAL=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FAMILY_STM32 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DMBED_BUILD_TIMESTAMP=1502268650.94 -DDEVICE_FLASH=1 -DTOOLCHAIN_ARM -DTOOLCHAIN_ARM_STD
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Misc      | 45240 |  2510 | 6156 |
| Subtotals | 45240 |  2510 | 6156 |
+-----------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 8666 bytes
Total RAM memory (data + bss + heap + stack): 8666 bytes
Total Flash memory (text + data + misc): 47750 bytes
Image: BUILD/tests/XDOT_L151CC/ARM/features/frameworks/utest/TESTS/unit_tests/test_setup_failure/test_setup_failure.bin

Building project mail (XDOT_L151CC, ARM)
Scan: ARM
Scan: mail
[DEBUG] Macros: -DTARGET_XDOT_L151CC -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32L151CC -DTOOLCHAIN_object -D__CMSIS_RTOS -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DDEVICE_RTC=1 -DDEVICE_ANALOGOUT=1 -DTARGET_STM32L1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_SERIAL=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FAMILY_STM32 - [... cut ...]
  File "c:\python27\lib\multiprocessing\util.py", line 325, in _exit_function
    p.join()
  File "c:\python27\lib\multiprocessing\process.py", line 145, in join
    res = self._popen.wait(timeout)
  File "c:\python27\lib\multiprocessing\forking.py", line 297, in wait
    res = _subprocess.WaitForSingleObject(int(self._handle), msecs)
KeyboardInterrupt
mbedgt: run_test_thread.run_host_test() failed, aborting...
mbedgt: ctrl+c keyboard interrupt!
[mbed] User aborted!

C:\Users\Gorazd\Desktop\xdit_novi_iz_armmbedos\mbed-os>

gorazdko, 2017-08-14:
Tests start to fail with 912c0de938c56528de88c11415b53b116dce6e5e  (#4740)
Can anyone else reproduce this?
@jeromecoutant 
@0xc0170

0xc0170, 2017-08-16:
cc @mfiore02

jeromecoutant, 2017-08-16:
Hi
Check the clock configuration, I maybe made some mistake with #4740 ?

Try to add this in the targets.json file :

    "config": {
        "hse_value": {
            "value": "24000000",
            "macro_name": "HSE_VALUE"
        }
    },

gorazdko, 2017-08-16:
Tnx @jeromecoutant. I was looking at the spot but wasnt sure why tests passed in #4740 by the mbed-bot.

commit 34bd3e5c (2017-08-24): xdot: fix target clock config in targets.json
Resolves #4876
closed 2017-08-24 by theotherjimmy