ARMmbed/mbed-os #4880: UBLOX_EVK_ODIN_W2 missing BUTTON1 definition

simosillankorva, 2017-08-09:
### Description
- Type: Bug
- Priority: Minor

## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
any

**mbed-os sha:**
4c256f0

**Expected behavior**
BUTTON1 expected to be defined.

**Actual behavior**
Not compiling if trying to use BUTTON1, it seems to be undefined.

0xc0170, 2017-08-09:
cc @andreaslarssonublox @andreaspeterssonublox

sg-, 2017-08-16:
Why do you think BUTTON1 should be defined?

closed 2017-08-16 by sg-

simosillankorva, 2017-08-17:
Is it not the standard name for the button nowdays? https://github.com/ARMmbed/mbed-os/commit/ad8f2fcbdf8e6e2941de896dfc7db4d74e4aaf9a

0xc0170, 2017-08-17:
> Is it not the standard name for the button nowdays? ad8f2fc

@kegilbert

kegilbert, 2017-08-17:
We had added in a set of standardized button names to match how we map LED pins to LED1...LEDn.

https://github.com/ARMmbed/mbed-os/pull/4249

This change hit around ~70 targets that had some form of button mapping but with an inconsistent set of names. There are still targets that have physical buttons but had no software definition for those buttons that will require going through and manually adding them.

sg-, 2017-08-19:
No tests, no docs, not the standard.

JanneKiiskila, 2017-08-21:
Why can't just fix this? It's no big deal to fix it and we have no intentions to start having the button definitions per board in the example client.

JanneKiiskila, 2017-08-21:
How hard would it be to add the porting guide info here; https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/porting_guide/ 

- it really should not be too hard.
- we should also demand they run [mbed-os-example-client](https://github.com/ARMmbed/mbed-os-example-client) in minimum.

sg-, 2017-08-21:
> Why can't just fix this?

Write some docs, create some tests and then start a review. That is the process for a 'feature request'. What is the defined behavior for boards that dont have any buttons?

JanneKiiskila, 2017-08-22:
I would recommend you @sg- take a look at the original PR then, as this thing is not "our thing", we just spotted one mistake - https://github.com/ARMmbed/mbed-os/pull/4249.

0xc0170, 2017-08-22:
> Write some docs, create some tests and then start a review. That is the process for a 'feature request'. What is the defined behavior for boards that dont have any buttons?

@kegilbert Can you please , if this has not been done (I noticed there was a request for docs in the original PR adding buttons) .

kegilbert, 2017-08-22:
I wrote up a quick blurb, need to run it by @AnotherButler