ARMmbed/mbed-os #4888: UDP MBED-OS 5.5.4 Static IP : Nucleo767ZI

danielklioc, 2017-08-11:
### Description
- Type: Bug 
- Related issue: #3524
- Priority: Major

---------------------------------------------------------------
## Bug
While sending or receiving data using UDP and Static IP, seems not happening anything on Ethernet socket. 

Tested with 5.5.3 and 5.5.2. Result: working, but after some time hangs and never starts to work again. 

**Target**
NUCLEO767ZI
NUCLEO746

**Also tested with online compiler**

**Offline IDE**
Eclipse Neon 2

**Toolchain:**
GCC_ARM

**Toolchain version:**
5_4-2016q3

`#include "mbed.h"
#include "EthernetInterface.h"

DigitalOut led1(LED1);
DigitalOut led2(LED2);

static const char*          mbedIP       = "192.168.0.48";  //IP  
static const char*          mbedMask     = "255.255.255.0";  // Mask
static const char*          mbedGateway  = "192.168.0.3";    //Gateway

static const char*          recvIP       = "192.168.0.38";  //IP 

//Network port to set-up socket address
const int PORT = 8051;

// ethernet network object on which will be build whole network with UDP socket
    EthernetInterface eth;

// create UDP socket
    UDPSocket socket(&eth);

void transmit(){
        
        //Buffer for output messages
        const char* out_buffer = "very important data";

        //address of destination ip address and port
        SocketAddress td(recvIP, PORT);

        // Loop to send data to Socket address "td"
       while(true){
            int ret = socket.sendto(td, &out_buffer, sizeof(out_buffer));
            //printf("sendto return: %d\n", ret); 
            Thread::wait(5000);
            }     
}

int main(){
    Thread transmitter;
   
        eth.set_network(mbedIP, mbedMask, mbedGateway); 
        eth.connect();
 
     transmitter.start(transmit);
}
`

**Expected behavior**
To get messages on second NUCLEO767ZI with IP address: 192.168.0.38
On Wireshark to see any communication between mentioned above IP`s or request from transmitter side for answer from second 
**Actual behavior**
Nothing happening and nothing seen on Wireshark side.

-----------------------------------------------------------------

## Question

**Is my example correct?**
**Is it something to do with Eclipse Neon 2 version?**
**Is it still issue not working Static IP?**
** Why it stops working on previous releases after few minutes?**
**Can transmitting thread run a same time with CAN IRQ?** 

This is my first issue report and I am beginner programmer, please tell me if something is  wrong or not clear.
Thank you

0xc0170, 2017-08-11:
cc @bcostm @adustm @jeromecoutant @LMESTM

danielklioc, 2017-09-28:
Is Nucleof767ZI working properly with UDP Static IP with mbed-os 5.5.6 & 5.5.7?
Is any documentation explaining how to use autoconfig(zeroconfig) ?

danielklioc, 2017-10-03:
Closed because issue is not clear and cannot be reproduced be others

closed 2017-10-03 by danielklioc