ARMmbed/mbed-os #4891: Add mbed version macros to mbed compile command.

nuket, 2017-08-11:
So the upgrade to CMSIS-RTOS2 in mbed 5.5 hosed the https://github.com/nuket/mbed-memory-status library I'd built. That's fine, since it was using a lot of low-level, non-public APIs.

But then I realized there isn't even a way to build a single translation unit compatible with versions < 5.5 and >= 5.5 because there are no mbed version macros passed to the compiler.

## Current Behavior

Let's say I'm using an NRF52_DK, this is what `mbed compile` passes as macros:

```
[DEBUG] Macros: -DDEVICE_ERROR_PATTERN=1 -DTARGET_MCU_NRF52832 -DTARGET_SDK11 -D__MBED__=1 -DNRF52_PAN_53 -DTARGET_NRF52_COMMON -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DBOARD_PCA10040 -DTARGET_FF_ARDUINO -DNRF52_PAN_12 -DTOOLCHAIN_GCC -DNRF52_PAN_58 -DDEVICE_SPISLAVE=1 -DNRF52_PAN_55 -DNRF52_PAN_54 -DTARGET_CORTEX_M -DNRF52_PAN_30 -DTARGET_NRF52_DK -DNRF52_PAN_36 -DTARGET_LIKE_CORTEX_M4 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NRF5 -DFEATURE_BLE=1 -DTARGET_MCU_NRF52 -DCMSIS_VECTAB_VIRTUAL -DNRF52_PAN_51 -DDEVICE_SPI_ASYNCH=1 -DNRF52 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DTARGET_NRF52832 -D__CORTEX_M4 -DMBED_BUILD_TIMESTAMP=1502461424.78 -DDEVICE_SERIAL=1 -D__FPU_PRESENT=1 -DS132 -DDEVICE_SLEEP=1 -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_NORDIC -DDEVICE_SERIAL_FC=1 -DBLE_STACK_SUPPORT_REQD -DNRF52_PAN_15 -DCONFIG_GPIO_AS_PINRESET -DTARGET_LIKE_MBED -D__MBED_CMSIS_RTOS_CM -DNRF52_PAN_31 -DTOOLCHAIN_GCC_ARM -DSWI_DISABLE0 -DSOFTDEVICE_PRESENT -DDEVICE_I2C_ASYNCH=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_INTERRUPTIN=1 -DNRF52_PAN_64 -DNRF52_PAN_20 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DNRF52_PAN_62 -DNRF52_PAN_63 -DDEVICE_SPI=1 -DARM_MATH_CM4
```

## Enhancement

Please add these macros:

```
-DMBED_VERSION_MAJOR=
-DMBED_VERSION_MINOR=
-DMBED_VERSION_PATCH=
```

So I can use the proper `#ifdef`s in my `.c` files.

0xc0170, 2017-08-11:
cc @adbridge @sg- @theotherjimmy

0xc0170, 2017-08-11:
These macros are in the mbed header file, it should be included to get this in. Is this a proposal to add global macros, and use those in the header file?

nuket, 2017-08-11:
Ok, I've tried that, but pulling in all of mbed just to get the versioning defines means that my `.c` file gets borked by transitive C++ includes.

```
Building project mms-test (NRF52_DK, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_BLE
Compile [  9.0%]: mbed_memory_status.c
[Fatal Error] Callback.h@21,15: new: No such file or directory
[ERROR] In file included from ./mbed-os/rtos/Thread.h:30:0,
                 from ./mbed-os/rtos/rtos.h:31,
                 from ./mbed-os/mbed.h:38,
                 from .\mbed-memory-status\mbed_memory_status.c:24:
./mbed-os/platform/Callback.h:21:15: fatal error: new: No such file or directory
 #include <new>
               ^
compilation terminated.
```

So there could be two options:
1. I switch my file to `.cpp` and extern "C" my public API functions.
2. `mbed.h` includes something like `mbed-version.h` which doesn't have this issue, which I can include in C.

There's also a question of dependency layering, if my code has nothing to do with the high-level C++ driver APIs, it probably shouldn't be including them, as they are higher-level API.

Also, a lot of headers aren't wrapped with `#ifdef __cplusplus`, which makes life fun.

Whether or not it's useful to keep supporting things for mbed OS versions lower than 5.5 is another question, of course. But the code's already been written, so I'd hate to throw it away.

0xc0170, 2017-08-14:
> Also, a lot of headers aren't wrapped with #ifdef __cplusplus, which makes life fun.

If any of these C header files are out there, we should fix those. Please report. If we are talking just about mbed.h, that one is C++.

> There's also a question of dependency layering, if my code has nothing to do with the high-level C++ driver APIs, it probably shouldn't be including them, as they are higher-level API.

Let's hear others what they think about having separate version header file (2nd option). Opinions anyone?

nuket, 2017-08-15:
IMHO, that would be ideal.

sg-, 2017-08-15:
What from mbed would be used that is not C++? It's the only public and supported API.

 ```#ifdef __cplusplus``` and ```extern "C"``` is the proper way to mix C and C++

nuket, 2017-08-16:
This, for instance -- https://github.com/nuket/mbed-memory-status

sg-, 2017-08-16:
I recall seeing that. Why is that not added as an extension to stack/heap stats that are already in place in mbed OS?

nuket, 2017-08-21:
At the time it was written, this was how it was written. Also, mbed OS had no means of tracking how much of the ISR stack was being used. And there seemed to be a number of different issues with the various memory size defines (STKSIZE, etc.) which I preferred to avoid, it was easier to grab the thread control block after the preprocessor finished its job, and introspect that.

There was no expectation on my side that the entire OS base layer would be replaced w/o having some means of conditionally compiling out low-level C code based on the mbed OS version.

deepikabhavnani, 2018-07-18:
Closing this issue: Resolved by PR #7434.

closed 2018-07-18 by deepikabhavnani