ARMmbed/mbed-os #4894: Pin missing from PeripheralPins.c

brandonboesch, 2017-08-11:
### Description
- Type: Bug  
- Priority: Major

---------------------------------------------------------------
**Target**
STM32F401 Nucleo

**mbed-os sha:**
f4864dc Merge pull request #3930 from ARMmbed/release-candidate

**Issue**
The issue is in regards to PeripheralPins.c for the STM32F401 Nucleo board which can be found at :
targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xE/TARGET_NUCLEO_F401RE/PeripheralPins.c

In this file, the pinmap for SPI_SSEL is missing its default arduino header pin that is defined in PinNames.h  The image below shows two different pin maps for the device from PeripheralPins.c, SPI_SCLK and SPI_SSEL.  Notice how SPI_SCLK has the arduino pin commented on, and that SPI_SSEL does not.

[image]

Now look at PinNames.h found at: /targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xE/TARGET_NUCLEO_F401RE/PinNames.h

You will notice that the default SPI_CS pin is PB_6 (as highlighted in the image below), which is not added to the PinMap in PerphiralPins.c.  I have not checked whether other STM Nucleo boards have similar issues as well.
[image]

brandonboesch, 2017-08-11:
@screamerbg

0xc0170, 2017-08-14:
cc @bcostm @adustm @LMESTM @jeromecoutant

jeromecoutant, 2017-08-16:
Hi
I agree this is misleading...
But I think there is no issue as in MBED, we are using SW NSS management.
So SPI_CS, connected to PB_6/D10, is used as a GPIO out.
The HW part (pins in PinMap_SPI_SSEL) are not really used...

As an example , you can see it in
https://github.com/ARMmbed/ci-test-shield/blob/master/TESTS/API/SPI/SPI.cpp#L54

LMESTM, 2017-08-16:
@brandonboesch - one more question, are you facing an actual issue ?
Physically on the board the SPI HW NSS is indeed not connected to the PB6/D10 pin, so you actually have 2 possibilities 
1) use SPI_CS as SW GPIO as done most of the time in MBED as explained by @jeromecoutant  
or 
2) use another pin from the list in PinMap_SPI_SSEL table if you want to use SPI HW NSS.

brandonboesch, 2017-08-16:
Thanks for the clarification.  Using SPI_CS as SW GPIO will work fine for my purposes.  Closing the issue.

closed 2017-08-16 by brandonboesch