ARMmbed/mbed-os #4897: How to change stack and heap size?

maxpeng, 2017-08-14:
If my board is NUCLEO_F429ZI and compiler is gcc, the stack and heap size are defined in `mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/device/TOOLCHAIN_GCC_ARM/STM32F429xI.ld` in mbed OS 5.x.

*Question:*
Is there a way to change the stack and heap size through mbed configuration system instead of modifying the liner configuration file of the target directly?

I tried to use a `mbed_app.json` file placed in the root directory of the project, but it didn't work.
```
{
    "target_overrides": {
        "*": {
            "target.stack_size": 4096,
            "target.heap_size": 65536
        }
}
```

0xc0170, 2017-08-14:
It is defined without being able to reconfigure it at the moment. As you can see the .ld has it statically defined via a specific number.

Can you provide a use-case for reconfiguring? 

- stack size in the ld results being used as the ISR stack
- heap size - only IAR has it currently limited that it is statically allocated the other 2 toolchains have it dynamic (using available space)

cc @theotherjimmy

maxpeng, 2017-08-14:
When I tried Azure IoT SDK for C and mbed TLS with IAR compiler, I found they required a large heap for dynamically memory allocation. Now I am trying to use GCC ARM compiler (this my [work](https://github.com/maxpeng/mbedos_iothub_client_sample_mqtt)), I mistakenly think I also need to configure the *.ld with a large heap.

I don't know GCC ARM and ARM compiler will dynamically adjust the heap until you mentioned that. I check the memory layout of *.ld of mbed OS 5.x, the stack is placed at the end of RAM, heap is placed before stack.
```
|--------|  ---> Start (lower address) of RAM
|        |
|        |
|--------|  
|        |  ---> Heap specified by *.ld
|  HEAP  |    | (grow upwards)
|        |    v
|--------|
|        |
|  Free  | 
|        |
|--------|  ---> lower address of RAM
| Stack  |    ^
|        |    |  (grow downwards) 
|--------|  ---> End of RAM
```
It seems the [`_sbrk`](https://en.wikipedia.org/wiki/Sbrk) will adjust the end of heap, when the existing heap space is used up, until it hits the stack region.

If my understanding is correct, then it should be no need to reconfigure heap size when we use GCC ARM and ARM compiler. However, it will be a bonus if we can re-configure heap size for IAR compiler, otherwise we need to modify the source code of mbed OS to meet our needs.

0xc0170, 2017-08-14:
> If my understanding is correct, then it should be no need to reconfigure heap size when we use GCC ARM and ARM compiler. However, it will be a bonus if we can re-configure heap size for IAR compiler, otherwise we need to modify the source code of mbed OS to meet our needs.

For IAR, only changing directly .icf file. We are hoping to have this fixed (IAR supports the same feature as GCC ARM or ARMCC) in the next releases of IAR (yet could not find this supported).

0xc0170, 2018-06-21:
I consider this question to be answered. Heap is dynamically sized for ARMCC/GCC ARM (IAR 8 has not yet landed).

If there is a need to add some configuration, please create a new ticket with feature request.

closed 2018-06-21 by 0xc0170