ARMmbed/mbed-os #4899: FAT filesystem causes constant corruption

LemonBoy, 2017-08-14:
I'm trying to use a very old 128mb microsd card with a modified [sd-driver](https://github.com/ARMmbed/sd-driver) to perform an alternative initialization sequence.
The card is initialized and reports a `SECTOR_SIZE` of `128` thus different from `block_size`: thats the main problem that prevents `FATFileSystem` to work, somehow `get_erase_size` is often used instead of `get_read_size` or `get_program_size` (eg. in `disk_read` or the implementation of `GET_SECTOR_SIZE`).
I've manually corrected those problems and removed the erase phase before the writing phase and got the I/O to work but somehow the filesystem gets corrupted after a while for no apparent reason.

0xc0170, 2017-08-14:
cc @geky @simonqhughes

geky, 2017-08-14:
Oh very interesting! What does your modified sd-driver report for the `get_read_size`, `get_program_size`, and `get_erase_size`?

Does if work with this patch? https://github.com/ARMmbed/mbed-os/pull/4843
I can clean that guy up and try to get him in if it does.

The FATFileSystem has a hard minimum of 512 bytes per sector. I don't think there are very many checks in place, so corruption is likely if the sector sizes are smaller. Fortunately, the FAT filesystem can just cluster the smaller sectors together if told to (which is what https://github.com/ARMmbed/mbed-os/pull/4843 does).

LemonBoy, 2017-08-15:
The driver reports `get_read_size = get_program_size = 512` but `get_erase_size = 128`.

geky, 2017-08-16:
Ah, one thing to note, the `get_read_size` and `get_program_size` should be smaller than `get_erase_size`. 

For SD cards, they should all be equal. These are just the physical properties of the card. The "logical sector size" is left up to the filesystem to decide. Currently the FATFileSystem doesn't let you set this easily, but I'm hoping to fix this in #4843.

Although 128 byte blocks is a strange number. Do you happen to have a link to information on the SD card by any chance?

LemonBoy, 2017-08-16:
While the controller is happy enough to use 512-bytes blocks after setting it with `CMD16` the erase size that's encoded in the CSD register reports `0x7F` (which reads as 128 according to the specification).
I think that forcing `_erase_size` to match `_block_size` is fine as long as the former divides the latter but that's something that needs to be changed in the sd-driver IMO.

> Although 128 byte blocks is a strange number. Do you happen to have a link to information on the SD card by any chance?

This is a old and small sd card, I guess that makes the block size less weird.

geky, 2017-08-16:
Agreed, looks like the sd-driver needs a patch: https://github.com/ARMmbed/sd-driver/pull/52

deepikabhavnani, 2017-08-16:
As per the SD specification, erase size of this card is 128*block size (512) which is 64K. Does the filesystem support that erase size?

geky, 2017-08-16:
Oh! that is very different. No, currently the FATFileSystem does not support block sizes > 4096.

Looking through the FATFileSystem implementation, you _might_ be able to get away with 64K blocks, but it will require a significant amount of modifications since the ChanFS hardcodes 32K as a maximum:
https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/fat/ChaN/ff.cpp#L4104-L4105

Unfortunately, I don't think we'll be able to support the FATFileSystem on that type of SD cards any time soon.

If you're really wanting to use that SD card in mbed OS, the [littlefs](https://github.com/ARMmbed/mbed-littlefs) might be able to do it. Although you will lose the ability to mount the SD card on your PC.

LemonBoy, 2017-08-17:
Alright, this explains everything I guess. I'd still like to have some `assert`s to check the implicit relations between the various `size` fields as it would've saved me some time (and many destroyed filesystems).

geky, 2017-08-17:
> The card works very well right now, I just removed the erase() call before the program() one (and worked around some other bugs in the sd-driver).

Oh, haha, I think that's the fastest I've been proven wrong.

If we can change the state of blocks on the SD card without an erase, the `get_erase_size` should just return the `get_program_size`, and `erase` should be a no op (or some sort of trim command). The `erase` function shouldn't limit the block size unless it is required as a part of writing new data. @deepikabhavnani

geky, 2017-08-17:
Looking into the SD card spec ([here](http://users.ece.utexas.edu/~valvano/EE345M/SD_Physical_Layer_Spec.pdf)), it looks like the erase command is just an optimization for future writes.

Since it's an optimization, we should only send the command if the range of bytes being erased fits. `get_erase_size` should still report 512, and the `erase` should just noop with <SECTOR_SIZE bytes being erased. If filesystems erase a larger range (at least spiffs does this), the optimization can kick in.

geky, 2017-09-12:
Closing as we think https://github.com/ARMmbed/sd-driver/pull/55 resolved the issue. Feel free to reopen if we're wrong.

closed 2017-09-12 by geky