ARMmbed/mbed-os #4901: Greentea fails for GCC_ARM in TMPM066

ganesh-ramachandran, 2017-08-14:
### Description
- Type: Bug
- Priority: Major | Minor

---------------------------------------------------------------
## Bug

**Target**
Toshiba TMPM066

**Toolchain:**
GCC_ARM

**Expected behavior**
Greentea testsuites (features-tests-filesystem-mbr_block_device & tests-mbedmicro-rtos-mbed-threads) should pass without change in main stack 

**Actual behavior**
Greentea testsuites (features-tests-filesystem-mbr_block_device & tests-mbedmicro-rtos-mbed-threads) fails due to not enough memory (Operator new[] out of memory). 

**Steps to reproduce**
1. Download master from https://github.com/ganesh-ramachandran/mbed-os (since Toshiba TMPM066 is yet to be merged)
2. mbed test -t GCC_ARM -m TMPM066 -v -n features-tests-filesystem-mbr_block_device,tests-mbedmicro-rtos-mbed-threads

----------------------------------------------------------------
## Details

Not enough memory is the message from the log - "Operator new[] out of memory" - Line numbers 191 & 289 from the attached log "M066_GCC_ARM_GreenTea_MainStack_4K.txt". Did not face this memory issue with ARM and IAR.

M066 Memory : RAM - 16K and Flash - 128K.

If main stack memory is reduced to 2K for GCC_ARM in mbed_rtx.h, then Greentea tests are passed (https://github.com/ARMmbed/mbed-os/pull/4840/commits/16af199332b91252c1d2514d8d94ba84cc3fb4f9).

Attached below the GCC_ARM logs with 4K and 2K main stack size.
[M066_GCC_ARM_GreenTea_MainStack_4K.txt](https://github.com/ARMmbed/mbed-os/files/1222319/M066_GCC_ARM_GreenTea_MainStack_4K.txt)
[M066_GCC_ARM_GreenTea_MainStack_2K.txt](https://github.com/ARMmbed/mbed-os/files/1222320/M066_GCC_ARM_GreenTea_MainStack_2K.txt)

Related Discussion are in #4840

0xc0170, 2017-08-14:
cc @bulislaw @c1728p9 @geky

bulislaw, 2017-08-15:
@0xc0170 do we have any 16k boards passing our testing?

0xc0170, 2017-08-15:
> @0xc0170 do we have any 16k boards passing our testing?

KL25Z has the same flash/ram sizes (not now plugged in CI though but it is released for mbed OS 5). NRF51 has even lower ram - 16k parts (16k - softdevice use) and it's passing. The tested nrf51dk has around 20k of RAM. We can look at those and analyze

cmonr, 2017-12-13:
Closing since fix looks to be merged in and issue inactivity.

closed 2017-12-13 by cmonr