ARMmbed/mbed-os #4904: Mutex 10005224 error -6: Not allowed in ISR context

chrissnow, 2017-08-14:
### Description
 Question

**Target**
LPC1768

## Question
Since updating mbed-os to 5.5 I'm getitng the following error
`Mutex 10005224 error -6: Not allowed in ISR context`

which makes sense, I'm using Mail in an ISR.

I'm using a serial RX interrupt to read in data, process a frame then stick it into a mailbox.

What's the best way of doing this?

chrissnow, 2017-08-14:
upon further investigation the cause actually seems to be calling time(NULL) which is odd...

chrissnow, 2017-08-15:
I will create a new issue for this as it's gong to be a mess here!

0xc0170, 2017-08-15:
@chrissnow All toolchains or just one ? I would guess GCC ARM? I'll look at the libc ,to me it seems time() invokation is protected by locks, that is causing this error. We had similar problem with mk time functions that we later replaced - as I recall rtc_read() was causing similar error to this one.

chrissnow, 2017-08-15:
@0xc0170 it will be all toolchains
time() makes use of mutexes which prevents it being used within an ISR, this doesn't seem right.
https://github.com/ARMmbed/mbed-os/blob/99a8467b70c5d6988f998514bcefc242642b5aa8/platform/mbed_rtc_time.cpp#L49

0xc0170, 2017-08-15:
Thanks, I did not realize that we overwrite time in retarget. we will look at this.

C99 standard does not specify if time and other related functions should be thread safe (it is stated for time conversion that they should - like `strftime` function), see section 7.23.2.4.

All our time functions implemented in retarged code file use a mutex.

> time() makes use of mutexes which prevents it being used within an ISR, this doesn't seem right.

What would you propose?

0xc0170, 2017-08-15:
cc @pan- @c1728p9 @bulislaw

chrissnow, 2017-08-15:
I'm no expert at where mutexes are going to be required here, but getting the current time shouldn't need one?

it looks like the RTC is initialised if it isn't already which then calls settime() which is then mutexed again.
```c++
  if (_rtc_isenabled != NULL) {
        if (!(_rtc_isenabled())) {
            set_time(0);
        }
    }
```

Happy to submit a PR if someone can suggest where mutexes are not needed etc.

bulislaw, 2017-08-15:
Yeah we can't lock mutexes from ISR obviously, do we know why there's a mutex? Is there a reason or just because other retargeted functions use them?

0xc0170, 2018-06-21:
This fixes it https://github.com/ARMmbed/mbed-os-5-docs/pull/361

closed 2018-06-21 by 0xc0170