ARMmbed/mbed-os #4906: IAR Export flags not synchronized

deepikabhavnani, 2017-08-14:
Below mentioned C++ flags are not set during export operation.
cxx_flags_cmd = ["--c++", "--no_rtti", "--no_exceptions"] 

IAR *.ewp fileversion 3, provides support to set separate C++ flags.
```
                <option>
                    <name>IccExceptions2</name>
                    <state>1</state>
                </option>
                <option>
                    <name>IccRTTI2</name>
                    <state>1</state>
                </option>
```
This support will require migration from template fileversion 2 to 3.

0xc0170, 2017-08-15:
> This support will require migration from template fileversion 2 to 3.

This is great that fileversion 3 provides C and C++ flags 🍾🎉  Is this new file version part of IAR 8.x ?

cc @theotherjimmy

deepikabhavnani, 2017-08-15:
Yes new version is part of IAR 8.x

deepikabhavnani, 2018-08-21:
Closing this as not in scope for now.

closed 2018-08-21 by deepikabhavnani