ARMmbed/mbed-os #4910: Small message types for Mail objects cause errors?

Maustemies, 2017-08-15:
### Description
- Type: Bug | Question

**Target**
NUCLEO_F429ZI

I am using a Mail object for one of my threads for inter-thread communications.
The message type struct basically contains two enums (result and command).   
```
typedef enum {
  MESSAGE_ERROR = -1,

  MESSAGE_OK = 0
} MessageResult;

typedef enum {
  COMMAND_ACTION_1,
  COMMAND_ACTION_2,
  COMMAND_ACTION_3
} MessageCommand;

typedef struct {
   MessageResult result;
   MessageCommand command;
} message_t;
```

I have been testing the functionality in the online compiler where my main() function is very simple as to just show that the program runs without problems:
```
static Mail<message_t, 10> mailBox;
int main() {
    
    int i = 0;
    while (1) {
        printf("asd %d\r\n", i++);
    }
    
    return 0;
}
```

However, when I compile the program (mbed online compiler), the program output to the serial port is:

> mbed assertation failed: _id, file: /extras/mbed-os.lib/rtos/MemoryPool.h, line 58

Now, when I add some padding to lengthen the message struct like this:
```
typedef struct {
   MessageResult result;
   MessageCommand command;
   union {
       bool val;
       uint16_t val2;
   } padding;
} message_t;
```
The program runs correctly with output:
```
asd 5966
asd 5967
asd 5968
asd 5969
asd 5970
asd 5971
...
```

Do you have any idea as to what causes this?
I have also tested this with the offline compiler using the GCC_ARM toolchain - same results, although the serial output is:
> Memory Pool 0x0 error -10

SenRamakri, 2017-08-21:
I'm able to reproduce this issue, looking into this.

bulislaw, 2017-08-22:
Should be fixed internally in https://github.com/ARMmbed/mbed-os/pull/4941

bulislaw, 2017-08-29:
Closing as #4941 was merged. Please reopen if you still have issues.

closed 2017-08-29 by bulislaw