ARMmbed/mbed-os #4914: Mbed OS  Architecture?

dadiliebian, 2017-08-15:
(1) What's Mbed OS Architecture?
(2) What's Mbed OS kernel Architecture?
As for question (2) , I think the mbed os kernel is uVisor,which is correct? Who tell me about the mbed os kernel ? thank you @MrJaba @hasselmm @julbouln @jrobeson @mpg

0xc0170, 2017-08-15:
@dadiliebian Can you clarify the questions? Have you looked at our doc pages, they should answer them : https://docs.mbed.com/

The above mentioned people do not look correct to me (not related to these questions at all).

dadiliebian, 2017-08-16:
oh,for example . Now I want to study the mbed OS and kernel. I think the uVisor is mbed os kernel as for Linux kernel one kind of Unix.So I want to konw uVisor and mbed os  releationship?

dadiliebian, 2017-08-16:
@0xc0170 thanks

0xc0170, 2017-08-16:
https://github.com/ARMmbed/uvisor/ - is hypervisor, not OS kernel. 

https://github.com/ARMmbed/mbed-os/tree/master/rtos/rtx5/TARGET_CORTEX_M - this is RTX5 that we use - http://www2.keil.com/mdk5/cmsis/rtx 

Browse our handbook as mentioned earlier: https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/cmsis-rtos

dadiliebian, 2017-08-16:
@0xc0170 understand ,thank you

closed 2017-08-31 by 0xc0170