ARMmbed/mbed-os #4915: Compiler error when building examples for NRF51_DK with GCC_ARM compiler

scartmell-arm, 2017-08-15:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
NRF51_DK

**Toolchain:**
GCC_ARM

When compiling `mbed-os-example-blinky` or `mbed-os-example-wifi` for NRF51_DK using GCC_ARM compiler it emits a compiler error.

**Steps to reproduce:**

```
$ mbed import mbed-os-example-blinky
$ cd mbed-os-example-blinky
$ mbed compile -m NRF51_DK -t GCC_ARM
```

Produces the error:

```
D:\mbed\mbed-os-example-wifi (master)
> mbed compile -m NRF51_DK -t GCC_ARM
Building project mbed-os-example-wifi (NRF51_DK, GCC_ARM)
Scan: .
Scan: env
Scan: mbed
Scan: FEATURE_BLE
Compile [ 70.2%]: rtx_mempool.c
[Error] ccBT9sRF.s@856,0: cannot honor width suffix -- `mov r1,#0'
[Error] ccBT9sRF.s@858,0: cannot honor width suffix -- `mov r0,#0'
[Error] ccBT9sRF.s@907,0: cannot honor width suffix -- `mov r4,#4'
[Error] ccBT9sRF.s@908,0: cannot honor width suffix -- `neg r1,r4'
[Error] ccBT9sRF.s@910,0: cannot honor width suffix -- `mov r5,#0'
[Error] ccBT9sRF.s@917,0: cannot honor width suffix -- `mov r7,#3'
[Error] ccBT9sRF.s@918,0: cannot honor width suffix -- `neg r1,r7'
[Error] ccBT9sRF.s@922,0: lo register required -- `add r0,r0,#12'
[Error] ccBT9sRF.s@982,0: cannot honor width suffix -- `mov r6,#4'
[Error] ccBT9sRF.s@984,0: cannot honor width suffix -- `neg r1,r6'
[Error] ccBT9sRF.s@986,0: cannot honor width suffix -- `mov r4,#4'
[Error] ccBT9sRF.s@993,0: cannot honor width suffix -- `mov r7,#3'
[Error] ccBT9sRF.s@994,0: cannot honor width suffix -- `neg r1,r7'
[Error] ccBT9sRF.s@996,0: cannot honor width suffix -- `mov r4,#3'
[Error] ccBT9sRF.s@998,0: cannot honor width suffix -- `neg r5,r4'
[Error] ccBT9sRF.s@1001,0: lo register required -- `add r0,r0,#12'
[Error] ccBT9sRF.s@1207,0: cannot honor width suffix -- `mov r2,#6'
[Error] ccBT9sRF.s@1208,0: cannot honor width suffix -- `neg r1,r2'
[Error] ccBT9sRF.s@1210,0: cannot honor width suffix -- `mov r0,#6'
[Error] ccBT9sRF.s@1211,0: cannot honor width suffix -- `neg r0,r0'
[ERROR] C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s: Assembler messages:
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:856: Error: cannot honor width suffix -- `mov r1,#0'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:858: Error: cannot honor width suffix -- `mov r0,#0'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:907: Error: cannot honor width suffix -- `mov r4,#4'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:908: Error: cannot honor width suffix -- `neg r1,r4'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:910: Error: cannot honor width suffix -- `mov r5,#0'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:917: Error: cannot honor width suffix -- `mov r7,#3'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:918: Error: cannot honor width suffix -- `neg r1,r7'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:922: Error: lo register required -- `add r0,r0,#12'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:982: Error: cannot honor width suffix -- `mov r6,#4'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:984: Error: cannot honor width suffix -- `neg r1,r6'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:986: Error: cannot honor width suffix -- `mov r4,#4'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:993: Error: cannot honor width suffix -- `mov r7,#3'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:994: Error: cannot honor width suffix -- `neg r1,r7'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:996: Error: cannot honor width suffix -- `mov r4,#3'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:998: Error: cannot honor width suffix -- `neg r5,r4'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:1001: Error: lo register required -- `add r0,r0,#12'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:1207: Error: cannot honor width suffix -- `mov r2,#6'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:1208: Error: cannot honor width suffix -- `neg r1,r2'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:1210: Error: cannot honor width suffix -- `mov r0,#6'
C:\Users\stecar01\AppData\Local\Temp\ccBT9sRF.s:1211: Error: cannot honor width suffix -- `neg r0,r0'

[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u D:\mbed\mbed-os-example-wifi\mbed-os\tools\make.py -t
CC_ARM -m NRF51_DK --source . --build .\BUILD\NRF51_DK\GCC_ARM" in "D:\mbed\mbed-os-example-wifi"
---
```

0xc0170, 2017-08-16:
@scartmell-arm This is known, we switched to GCC 6 since mbed OS 5.5 . 

As these errors are quite criptic, we might consider having some toolchains check in the mbed-cli or mbed-os code base? 

cc @theotherjimmy

theotherjimmy, 2017-08-16:
> As these errors are quite criptic, we might consider having some toolchains check in the mbed-cli or mbed-os code base?

I'm just confused by this question. It's phrased like a statement. What are you asking for?

0xc0170, 2017-08-17:
@theotherjimmy It could be a question. If when we update major versions of our toolchain and we are aware of incompabilities , we should add a check (warn a user?) so these type of errors are easily solved.

Anyway, this can be closed, as it's not valid (using old toolchain that is incompatible with the current codebase).

closed 2017-08-17 by 0xc0170