ARMmbed/mbed-os #4917: Plans for new Event Recorder system

ohagendorf, 2017-08-15:
### Description
- Type: ~~Bug | Enhancement |~~ Question
- ~~Related issue: `#abc`~~
- Priority: ~~Blocker | Major |~~ Minor

## Question

With the new included RTX5 a new debug possibility was included with it. There exists an event system in the RTOS kernel which can be easily enabled. Currently there seems to exist only the Keil µVision IDE with an Event Recorder, displaying these events during a debug session. And add-on tools like the Tracealyzer. But maybe additional tools will be submitted in the future.

I tested the possibility to use this feature in mbed and it works very well. And I tried to describe the usage in an article: https://developer.mbed.org/users/ohagendorf/notebook/mbed-rtx5-and-uvision-event-recorder/

Are there any plans to include this system natively into mbed? I think, creating events similar to RTX5 in other libraries as already done in the Keil middleware libraries, would improve the debug possibilities substantially.

0xc0170, 2017-08-16:
cc @bulislaw @c1728p9 @theotherjimmy

bulislaw, 2017-08-16:
I wouldn't call it a plan yet, but there's a wish :) We recognize possible usefulness of the system, but right now we just don't have the capacity to do much about it.

c1728p9, 2017-08-16:
cc @deepikabhavnani

deepikabhavnani, 2017-11-29:
@0xc0170 - I believe the question is answered (no plans at present), should we keep it open as enhancement in future, or close the question?

0xc0170, 2017-11-29:
@ohagendorf Thanks for the article and the proposal. The article is good, definitely help few of us. 

As above, we will consider this in the future, but currently not planned. Therefore as proposed by @deepikabhavnani we close this one for now.

closed 2017-11-29 by 0xc0170