ARMmbed/mbed-os #4930: MBED2 tests won't run after merge of #4392

LMESTM, 2017-08-17:
### Description
- Type: Bug 
- Related issue: `#4392`
- Priority: Blocker 

Since merge of #4392, we can't start MBED2 tests anymore.
The test seems to even not be compiled, nor run.

**BEFORE**
```
$ python tools/singletest.py --auto  -n MBED_A28 -V
[...]
Library: mbed.ar
Copy: mbed_board.o
Copy: mbed_retarget.o
Building project can_loopback (NUCLEO_F446RE, ARM)
Scan: can_loopback
Scan: mbed
Scan: env
Compile [ 50.0%]: main.cpp
Compile [100.0%]: test_env.cpp
Link: can_loopback
Elf2Bin: can_loopback
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Misc      | 27642 |    97 | 1672 |
| Subtotals | 27642 |    97 | 1672 |
+-----------+-------+-------+------+
Allocated Heap: unknown
Allocated Stack: unknown
Total Static RAM memory (data + bss): 1769 bytes
Total RAM memory (data + bss + heap + stack): 1769 bytes
Total Flash memory (text + data + misc): 27739 bytes

Looking for NUCLEO_F446RE with MBEDLS
Executing 'python host_test.py -d D: -f "C:\Data\Workspace\mbed\BUILD\test\NUCLEO_F446RE\ARM\MBED_A28\can_loopback.bin" -p COM68 -t 30 -C 1.5 --auto -m NUCLEO_F446RE'
Test::Output::Start
MBED: Instrumentation: "COM68" and disk: "D:"
HOST: Copy image onto target...
        1 file(s) copied.
HOST: Initialize serial port...
Looking for NUCLEO_F446RE with MBEDLS
etc ...
```
**AFTER**
```
$ python tools/singletest.py --auto  -n MBED_A28 -V
[...]
Library: mbed.ar
Copy: mbed_board.o
Copy: mbed_retarget.o
Test summary:
+--------+---------------+-----------+----------+-------------------+--------------------+---------------+-------+
| Result | Target        | Toolchain | Test ID  | Test Description  | Elapsed Time (sec) | Timeout (sec) | Loops |
+--------+---------------+-----------+----------+-------------------+--------------------+---------------+-------+
| FAIL   | NUCLEO_F446RE | ARM       | MBED_A28 | CAN loopback test |         0          |       0       |   -   |
+--------+---------------+-----------+----------+-------------------+--------------------+---------------+-------+
Result: 1 FAIL
Completed in 33.57 sec
Build successes:
  * NUCLEO_F446RE::ARM::MBED
```

**Target**
ALL

**Toolchain:**
ALL ?

**Steps to reproduce**
Use command
$ python tools/singletest.py --auto  -n MBED_A2 -V

LMESTM, 2017-08-17:
cc @theotherjimmy @MarceloSalazar @0xc0170

0xc0170, 2017-08-18:
Thanks for the report, I am looking at this now

0xc0170, 2017-08-18:
Found an error on this line https://github.com/ARMmbed/mbed-os/pull/4392/commits/7053ef9412c913e6ebedf0aa25049f8a5f371dc7#diff-055d1aa4e49070169f49841a86e1b6a4R500

PR #4933 (2017-09-04): test: fix stats depth variable scope
As this variable is not in scope, should not be referenced.
~Fixes #4392~ No, you can't do "fixes PR#" only "fixes issue#" makes sense
Fixes #4930 

~I am not certain this is a proper fix. The variable is not in the scope in execute() . Should it be passed somehow there, or just ignored ? This fixes it as we are not using undefined variable. The question is how it should be passed to execute function or should not and this is correct?~
@theotherjimmy update: I passed stats depth correctly.

Tested with singletest.py. With this patch, it builds. Without, bui [... cut]
commit b99e6b5f (2017-09-04): Pass stats depth to singletest correctly

Fixes #4930
commit 03e45f40 (2017-09-12): Pass stats depth to singletest correctly

Fixes #4930
commit f50ce451 (2017-09-13): Pass stats depth to singletest correctly

Fixes #4930
closed 2017-09-04 by 0xc0170