ARMmbed/mbed-os #4942: Update e2 studio exporter

phyokyaw, 2017-08-21:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question
- Related issue: `#abc`
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Enhancement

**Target**
RZ-A1H

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
1.2.0

**mbed-os sha:**
ff4fc34

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

https://developer.mbed.org/blog/entry/Fixed-exporters-in-the-mbed-Online-Compi/

@theotherjimmy 
Hi Jimmy, I am looking into changing the e2 studio exporter to remove legacy code and extend from new Exporter class. Since e2 studio uses gnuarmeclipse toolchain the exporter would be almost identical to gnuarmeclipse exporter.

However we extend build configuration ID and toolchain ID, hence minor differences in the exporter .cproject template.

Will it be feasible to create exporter that extends gnuarmeclipse exporter class? We can override the generate() method to look for e2 studio templates. If it is possible to refactor generate() from gnuarmeclipse to move gen_file(...) out we only need to override that method.

Please let me know and I'll create a pull request. 

Thanks
Phyo Kyaw

phyokyaw, 2017-08-23:
Added a PR. Thanks.

0xc0170, 2018-06-26:
Resolved via #4959

closed 2018-06-26 by 0xc0170