ARMmbed/mbed-os #4946: 5.5.5 - CMSIS-RTOS: ISR Queue Overflow or Assetion Failed  while using Events

maiorfi, 2017-08-21:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F103RB | NUCLEO_F091RC

**Toolchain:**
ARM

**Toolchain version:**
5.24.2

**mbed-cli version:** 
1.2.0

**Expected behavior**
Blinking when processing uart rx isr

**Actual behavior**
Two different (bad) behaviors, depending on Event Queue Size: with small size (e.g. 4) "mbed assertation failed: id, file: .\mbed-os\events/Event.h, line 149", with bigger sizes (e.g. 32) "CMSIS-RTOS error: ISR Queue overflow (status: 0x2, task ID: 0x0, object ID: 0x2000206C)"

**Steps to reproduce**

```c++
#include "mbed.h"

static void rx_isr();
static void loop_fn();

static DigitalOut led4(LED3);
static DigitalOut led2(LED4);

static RawSerial uart_stdio(USBTX,USBRX,115200);

EventQueue queue(4 * EVENTS_EVENT_SIZE);

Thread eventLoopThread;
Thread mainLoopThread;

int main()
{
	uart_stdio.attach(queue.event(rx_isr));
	
	uart_stdio.puts("Blinky started...\n");
	
	eventLoopThread.start(callback(&queue, &EventQueue::dispatch_forever));
	mainLoopThread.start(loop_fn);
}

static void loop_fn()
{
	while (true)
    {
        led4 = !led4;
        wait_ms(500);
    }
}

static void blink_once()
{
	led2=true;
	wait_ms(250);
	led2=false;
	wait_ms(125);
}

static void rx_isr()
{
	queue.call(blink_once);
}

```

LMESTM, 2017-08-22:
@maiorfi  Ok so some feeback.
First note : I had to tune the main and thread stack sizes in mbed_app.json otherwise the example wouldn't start.
Once done, I can indeed reproduce this issue. 
Maybe the issue was not visible before 5.5.5. because there has been a fix in #4734 integrated in this version - without this fix you would possible loss data in application. The consequence of this fix is that the application needs to read the uart data when it has registered for the Rx ISR. So even if you discard it, you need to add a call to  uart_stdio.getc();

I have a modified example working fine as below:
```
#include "mbed.h"

static void rx_isr();
static void loop_fn();

static DigitalOut led4(LED3);
static DigitalOut led2(LED4);

static RawSerial uart_stdio(USBTX,USBRX,115200);

EventQueue queue(4 * EVENTS_EVENT_SIZE);

Thread eventLoopThread;
Thread mainLoopThread;

static void blink_once()
{
	led4=true;
	wait_ms(250);
	led4=false;
	wait_ms(125);
}

/*  under ISR context */
void rx_isr() {
    uart_stdio.getc();
    queue.call(blink_once);
}

int main()
{
	uart_stdio.puts("Blinky started...\n");
	uart_stdio.attach(&rx_isr);
	
	eventLoopThread.start(callback(&queue, &EventQueue::dispatch_forever));
	mainLoopThread.start(loop_fn);
}

static void loop_fn()
{
	while (true)
    {
        //led4 = !led4;
        wait_ms(500);
    }
}
```
Hope this helps ...

maiorfi, 2017-08-23:
@LMESTM That's fine: now it works :)

It turns out that issue is actually caused by using

```c++
uart_stdio.attach(queue.event(rx_isr));
```

instead of using

```c++
uart_stdio.attach(rx_isr);
```

or (that should be treated by compiler in the same way as above):

```c++
uart_stdio.attach(&rx_isr);
```
 
Interesting enough, I do not need to specify a mbed_app.json config file (maybe it depends on compiler? I am using Keil/ARM).

Should I need it, would following be correct, according to your suggestion?

```json
{
  "config": {
    "thread_stack_size": {
      "value": 1024
    }
  }
}
```

Thanks!

closed 2017-08-23 by maiorfi