ARMmbed/mbed-os #4952: UBLOX_EVK_ODIN_W2 sporadic ethernet socket connect failures

avolinski, 2017-08-22:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
## Bug

**Target**
UBLOX_EVK_ODIN_W2

**Toolchain:**
GCC_ARM,ARM

**Toolchain version:**
GCC_ARM 6

**mbed-os sha:**
mbedOS 5.5.5

**Expected behavior**
Ethernet socket connect always succeeds

**Actual behavior**
Ethernet socket connect sporadically fails

**Steps to reproduce**
Do lot's of connects and disconnects, with small random delay between them
----------------------------------------------------------------

andreaslarssonublox, 2017-08-24:
@avolinski have you tested it with any other board like STM32F429 Nucleo that has the same ethernet drivers?

0xc0170, 2017-08-31:
> @avolinski have you tested it with any other board like STM32F429 Nucleo that has the same ethernet drivers?
 
This would be good to know. 

@andreaslarssonublox  you supspect that this is F4 network driver problem rather than just reported target?

avolinski, 2017-10-02:
Hi,
We only test ODIN, NUCLEO_F429ZI and NXP K64F boards

andreaslarssonublox, 2017-10-18:
[main.zip](https://github.com/ARMmbed/mbed-os/files/1394344/main.zip)
Hi,
I have tested to connect/disconnect ethernet 100 times and it succeeded for all. Also tried to create/connect/close a TCP socket 100 times. It also worked so maybe I have misunderstood the use case. Can you send some example where we can see the problem?

I have attached a test example where I have tried to reproduce.

0xc0170, 2018-06-26:
Please reopen with an update (@andreaslarssonublox could not reproduce the issue).

closed 2018-06-26 by 0xc0170