ARMmbed/mbed-os #4957: How to call spi_free()

janjongboom, 2017-08-22:
I need to call `spi_free()` because my SPI flash chip does not like to be init'ed from both bootloader and app.

But it's not available in the C++ API... Any idea how to do it?

JojoS62, 2017-08-22:
I doubt that it will help because I've seen only empty implementations. And the comment in hal/spi_api.h says:
```
/** Release a SPI object
 *
 * TODO: spi_free is currently unimplemented
 * This will require reference counting at the C++ level to be safe
 *
 * Return the pins owned by the SPI object to their reset state
 * Disable the SPI peripheral
 * Disable the SPI clock
 * @param[in] obj The SPI object to deinitialize
 */
void spi_free(spi_t *obj);
```

0xc0170, 2017-08-31:
This will require further work, as noted above, in the documentation, needs specification and implementation for targets.

janjongboom, 2017-09-04:
@0xc0170 I'd suggest adding a `free()` function to SPI object. Perhaps we'd mark it in the docs with an indication that you probably never want to call it.

janjongboom, 2018-06-02:
Just in case someone else runs into this problem, I made a DestructableSPI class which can free the interface: https://github.com/janjongboom/at45-blockdevice/blob/44766342130b373f59ac372add62e73f55469f9b/DestructableSPI.h

ithinuel, 2018-06-26:
This is also addressed in the [[WIP] SPI hal update](https://github.com/ARMmbed/mbed-os/pull/6782).

deepikabhavnani, 2018-09-11:
Issue is addressed in SPI hal, can this be closed now?

janjongboom, 2018-09-12:
@deepikabhavnani Where was the SPI Hal work landed?

deepikabhavnani, 2018-09-12:
Ready to merge https://github.com/ARMmbed/mbed-os/pull/7671 - Feature branch `feature-hal-spec-spi`.

janjongboom, 2018-09-12:
@deepikabhavnani That's just design documents, where is the actual work? Unless there's some actual work being done we cannot close this.

ithinuel, 2018-09-12:
There are work on reference implementations for some ST's and Freescale's targets.
They'll be part of an upcomig PR to the same feature branch.

deepikabhavnani, 2018-09-18:
Duplicate of https://github.com/ARMmbed/mbed-os/issues/3106.
Please use the Issue: 3106 to track anything related to destructor, free pins, disable module / driver.. etc

closed 2018-09-18 by deepikabhavnani