ARMmbed/mbed-os #4967: FlashIAP API flash.init() fail when called second time

LiyouZhou, 2017-08-23:
https://github.com/ARMmbed/mbed-os/blob/master/drivers/FlashIAP.cpp#L56

FlashIAP is designed to be a singleton. Hence FlashIAP::init() should be able to be called multiple times from multiple places without issue.

The current implementation exposes the HAL behaviour to the user and behaves differently on different platforms. K64F would succeed while STM32 would fail with multiple calls to FlashIAP::init()

0xc0170, 2017-08-31:
cc @LMESTM is https://github.com/ARMmbed/mbed-os/pull/4980 also fixing this issue?

LMESTM, 2017-08-31:
@0xc0170  I haven't tried that yet ... will have a look soon

LMESTM, 2017-10-10:
@LiyouZhou Your problem should have been solved with #5026 - if confirmed, can you please close this issue ?

0xc0170, 2017-10-12:
Reopen if this is still valid

closed 2017-10-12 by 0xc0170