ARMmbed/mbed-os #4969: SPI format and frequency may not be set in SPI constructor

ccchang12, 2017-08-24:
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

The SPI class has a member "static SPI *_owner;" to indicate need to set SPI or not, and the default value is NULL in SPI.cpp. But after call the destructor and constructor again, it may encounter the format and frequency are not set issue due to the address is the same.

In ci-test-shield\TESTS\API\SPI\SPI.cpp, each test case calls the SPI constructor and may cause this issue. The solution should be set the _owner to be NULL in the SPI destructor as below.

diff --git a/drivers/SPI.h b/drivers/SPI.h
index 645243070..7e0ba08f4 100644
--- a/drivers/SPI.h
+++ b/drivers/SPI.h
@@ -263,6 +263,7 @@ protected:

 public:
     virtual ~SPI() {
+        _owner = NULL;
     }

 protected:

0xc0170, 2017-08-31:
cc @deepikabhavnani @c1728p9 @bulislaw

bulislaw, 2017-08-31:
@ccchang12 Do you mind pushing PR with a fix?

ccchang12, 2017-09-01:
@bulislaw I think it is better the owner can modify the code directly in next release.

closed 2018-10-05 by 0xc0170