ARMmbed/mbed-os #4970: UBLOX_C027 - 'Flash - buffer alignment test' is failing

fahim-ublox, 2017-08-24:
### Description
- Type: Bug
- Related issue: `#4288`
- Priority: Minor

**Target**
UBLOX_C027 

**mbed-os sha:**
(`316c875 Merge pull request #4926 from bcostm/merge_l053x8`)

**Expected behavior**
Flash - buffer alignment test should pass for UBLOX_C027 

**Actual behavior**
'Flash - buffer alignment test' is failing for UBLOX_C027 

**Steps to reproduce**
mbed test -m UBLOX_C027 -t GCC_ARM 

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
As test for UBLOX_C027  is failing  and it was reported for other target as well. Test was disabled for other target as per #4288. If you like to add more target(UBLOX_C027 ) to disable list until test is fixed ? or disable it for all targets 

@0x6d61726b  @0xc0170

chrissnow, 2017-08-29:
@fahim-ublox  I have linked on #4288 my WIP for IAP, I ran out of time to finish it but will hopefully be back on it soon, I don't think it will take much to finish it.

chrissnow, 2017-08-30:
#4993 Should resolve this.

0xc0170, 2017-08-31:
@chrissnow Thanks for the help here

closed 2017-12-16 by sg-